LibUsbDotNet/LibUsbDotNet #171: About the DetachKernelDriver for Linux systems

GreenShadeZhang, 2022-10-14:
I want to commit a pr in the v2 branch, but I see that v2 has been migrated to .net5, so are there plans to upgrade to .net6? Or do I fork a repository myself, and then modify and publish the nuget package on my own branch? Can someone answer my question? Unbinding via udev doesn't seem to take effect, so I want to call the DetachKernelDriver code directly.

mcuee, 2022-10-16:
@GreenShadeZhang 
Please help to fork the repo and create a pull request. Thanks.

Let's see if @qmfrederik has some time to review the pull requests.

GreenShadeZhang, 2022-10-16:
> @GreenShadeZhang Please help to fork the repo and create a pull request. Thanks.
> 
> Let's see if @qmfrederik has some time to review the pull requests.

I made a commit of the PR, you can test it over there, I used some modified code in one of my projects yesterday, the address is as follows.

https://github.com/maker-community/ElectronBot.DotNet

mcuee, 2023-01-06:
PR #172 has been merged into v2 branch.
* #172

GreenShadeZhang, 2023-01-07:
> PR #172 has been merged into v2 branch.
> 
> * [The project framework was upgraded to .NET 6 to add method calls for DetachKernelDriver and SetAutoDetachKernelDriver. #172](https://github.com/LibUsbDotNet/LibUsbDotNet/pull/172)

Can I release a NuGet package for the code merged from the v2 version? I want to modify the dependencies in my project to use the package you released

mcuee, 2023-01-08:
> Can you release a NuGet package for the code merged from the v2 version? I want to modify the dependencies in my project to use the package you published

@qmfrederik 
Just wondering if you can help to release a new version for the v2 branch. Thanks.

@GreenShadeZhang 
You may also try to contact the owners of nuget libusbdotnet with the info here.
https://www.nuget.org/packages/LibUsbDotNet/

mcuee, 2023-01-08:
BTW, for the master branch, there is a PR #165 to address this issue.
* #165

GreenShadeZhang, 2023-01-08:
> @qmfrederik Just wondering if you can help to release a new version for the v2 branch. Thanks.
>
> @GreenShadeZhang You may also try to contact the owners of nuget libusbdotnet with the info here. [...]

I saw the nuget package file you posted, why not upload it to the nuget website for public release? Is it because you don't have permission to operate?

mcuee, 2023-01-08:
> I saw the nuget package file you posted, why not upload it to the nuget website for public release? Is it because you don't have permission to operate?

That is correct. I am not the owner of the libusbdotnet in nuget.org.

GreenShadeZhang, 2023-01-09:
> > I saw the nuget package file you posted, why not upload it to the nuget website for public release? Is it because you don't have permission to operate?
> 
> That is correct. I am not the owner of the libusbdotnet in nuget.org.

So that's the case, then expect someone with permission to operate it.

mcuee, 2023-01-09:
> So that's the case, then expect someone with permission to operate it.

Yes, I believe @qmfrederik is one of the owners.

@qmfrederik Please help here. Thanks.

GreenShadeZhang, 2023-02-14:
> > So that's the case, then expect someone with permission to operate it.
> 
> Yes, I believe @qmfrederik is one of the owners.
> 
> @qmfrederik Please help here. Thanks.

Has the author been too busy lately, can he give you a permission?

mcuee, 2023-04-22:
PR https://github.com/LibUsbDotNet/LibUsbDotNet/pull/165 and PR https://github.com/LibUsbDotNet/LibUsbDotNet/pull/212 have been merged for the master branch.

closed 2023-04-22 by mcuee