LibUsbDotNet/LibUsbDotNet #177: Usb mutex lock

vitalijusb85, 2022-11-23:
Hi,  I have two connected devices. Through the loop, I trying open the device and clime interface. So sometimes I getting an error: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed. Aborted.
This error is breaking my app.
What am I doing wrong? 

Thanks for your help.

mcuee, 2022-11-24:
libusb-1.0.26 should be used. Most likely you have a bug in your application.

vitalijusb85, 2022-11-30:
Hi, you were right. It was an error in the application in switching different contexts. 
I have one question. Is it possible to set the baud rate? 
Thanks.

mcuee, 2022-11-30:
> Hi, you were right. It was an error in the application in switching different contexts. I have one question. Is it possible to set the baud rate? Thanks.

There is no such thing as baud rate for USB. If youa are talking about USB to Serial adapters, please do not use libusbdotnet. You need to use other libraries for that.

closed 2022-12-09 by mcuee

0wwafa, 2025-02-13:
I think libusb is bugged.
I am using libusb-1.0.25.
My program does high speed transfer from usb. Everything works but when at the end I issue the libusb close I get the same error as the OP.