LibUsbDotNet/LibUsbDotNet #178: Reading from port byte by byte

vitalijusb85, 2022-12-09:
Hi, 
in my case, I need to read data from port bytes by bytes. I need to analyze the frame.
First, I used the method:
 `public virtual Error Read(byte[] buffer, int offset, int count, int timeout, out int transferLength)`.
I know that my response is 45 bytes. But I want to read the first 6 bytes.  So my buffer size is 6 and offset 0 and count 6.
I get no response from the port. If I set a buffer length of 45 bytes or more it works.
Why doesn't it work?

Thanks for your help.

mcuee, 2022-12-09:
You have set the buffer as 45 bytes in this case. I believe that is the way libusb works.

closed 2022-12-09 by mcuee

vitalijusb85, 2022-12-09:
Thanks for the fast response. It is possible to get from the port buffer length?