LibUsbDotNet/LibUsbDotNet #181: win-x64 libusb-1.0.dll from nuget package causes DllNotFoundException

MatthewPlayer, 2023-01-06:
I'm using 3.0.102-alpha package from nuget and building for win-x64. 

The libusb-1.0.dll gets placed correctly but results in a DllNotFoundException. If I replace the x64 version of libusb-1.0.dll  with the x86 bit version you get a BadImageFormatException as expected but that proves that it is looking in correct place but the x64 version doesn't load for some reason.

I downloaded the nupkg file and extracted the dlls and reran the same test above and got the same result. I did manage to get it working with a libusb-1.0.dll file I copied from my dev PC in a msys64 directory.

BlueFinBima, 2023-01-06:
The same for me (same environment)  but not restricted to Alpha 3.0.102.  .Net Core builds work fine, but .Net Framework has the problem.  As mentioned in another issue, renaming `libusb-1.0.dll` to `libusb-1.0` allows it to be found, however I then get a BadImageFormatException  or Memory Access exception. 
@MatthewPlayer any chance you could point me towards the libusb-1.0.dll that you got to work?

mcuee, 2023-01-08:
@BlueFinBima 

What if you use the libusbdotnet nuget package and libusb-1.0.dll from github action?
https://github.com/LibUsbDotNet/LibUsbDotNet/actions/runs/3345889694

nuget package:
https://github.com/LibUsbDotNet/LibUsbDotNet/suites/9013914142/artifacts/415580862

libusb binary for different OS
https://github.com/LibUsbDotNet/LibUsbDotNet/suites/9013914142/artifacts/415580860

mcuee, 2023-01-08:
This is the build log.

```
PS C:\work\libusb\LibUsbDotNet> dotnet pack src\LibUsbDotNet\LibUsbDotNet.csproj -c Release
MSBuild version 17.3.2+561848881 for .NET
  Determining projects to restore...
  All projects are up-to-date for restore.
  LibUsbDotNet -> C:\work\libusb\LibUsbDotNet\src\LibUsbDotNet\bin\Release\net45\LibUsbDotNet.dll
  LibUsbDotNet -> C:\work\libusb\LibUsbDotNet\src\LibUsbDotNet\bin\Release\netcoreapp3.1\LibUsbDotNet.dll
C:\Program Files\dotnet\sdk\6.0.404\Sdks\NuGet.Build.Tasks.Pack\buildCrossTargeting\NuGet.Build.Tasks.Pack.targets(221,5): warning NU5048: The 'PackageIcon
Url'/'iconUrl' element is deprecated. Consider using the 'PackageIcon'/'icon' element instead. Learn more at https://aka.ms/deprecateIconUrl [C:\work\libus
b\LibUsbDotNet\src\LibUsbDotNet\LibUsbDotNet.csproj]
  Successfully created package 'C:\work\libusb\LibUsbDotNet\src\LibUsbDotNet\bin\Release\LibUsbDotNet.3.0.102-alpha.nupkg'.
  Successfully created package 'C:\work\libusb\LibUsbDotNet\src\LibUsbDotNet\bin\Release\LibUsbDotNet.3.0.102-alpha.snupkg'.

```

BlueFinBima, 2023-01-08:
@mcuee Many thanks for getting back to me.  I cleaned the NuGet cache and installed the package which I downloaded from the action, from a local directory.  I then tested with the libusb1.0.dll from the package and then the dll from the `x64-windows\bin` directory in  `libusb-3345889694.zip`.  As far as I could determine, the problem remained the same.

Specifically, I have two projects (with the same application code) both tested on Windows x64, the working one is a Console Application targeting .Net 6.0 and this works correctly with 3.0.102.alpha (both from the action and from NuGet).
The project exhibiting the problem, is  a "Console App (.NET Framework)" targeting .NET Framework 4.7.2.
### Test 1:   Built with Locally Installed NuGet Package from the action
```
Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'libusb-1.0': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at LibUsbDotNet.NativeMethods.Init(IntPtr& ctx)
   at LibUsbDotNet.LibUsb.UsbContext..ctor()
   at Examples.USBTesting.Main(String[] args) in E:\users\larmour\Source\Repos\BlueFinBima\USBTest\USBTest472\Program.cs:line 22
``` 
```
28/10/2022  13:46           173,056 libusb-1.0.dll
28/10/2022  13:47            53,248 LibUsbDotNet.dll
19/07/2017  10:01            27,992 System.Buffers.dll
06/06/2018  13:29           148,624 System.Memory.dll
19/07/2017  10:01           115,936 System.Numerics.Vectors.dll
15/05/2018  13:29            22,160 System.Runtime.CompilerServices.Unsafe.dll
08/01/2023  08:27             5,632 USBTest472.exe
05/01/2023  12:10               189 USBTest472.exe.config
08/01/2023  08:27            22,016 USBTest472.pdb
```
### Test 2:   Built with Locally Installed NuGet Package from the action and libusb-1.0.dll **renamed** to libusb-1.0
```
Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at LibUsbDotNet.NativeMethods.UnrefDevice(IntPtr dev)
   at LibUsbDotNet.Device.ReleaseHandle()
   at System.Runtime.InteropServices.SafeHandle.InternalFinalize()
   at System.Runtime.InteropServices.SafeHandle.Finalize()
```
```
28/10/2022  13:46           173,056 libusb-1.0
28/10/2022  13:47            53,248 LibUsbDotNet.dll
19/07/2017  10:01            27,992 System.Buffers.dll
06/06/2018  13:29           148,624 System.Memory.dll
19/07/2017  10:01           115,936 System.Numerics.Vectors.dll
15/05/2018  13:29            22,160 System.Runtime.CompilerServices.Unsafe.dll
08/01/2023  08:27             5,632 USBTest472.exe
05/01/2023  12:10               189 USBTest472.exe.config
08/01/2023  08:27            22,016 USBTest472.pdb
``` 
### Test 3:   Built with Locally Installed NuGet Package from the action and libusb-1.0.dll from the action libusb-3345889694.zip \x64-windows\bin (timestamp 13:45)
Same results as Test 1
### Test 2:   Built with Locally Installed NuGet Package from the action and libusb-1.0.dll from the action libusb-3345889694.zip \x64-windows\bin (timestamp 13:45) **renamed** to libusb-1.0
Same results as Test 2

mcuee, 2023-01-08:
Here is the zip for the MSYS2 mingw64 version of libusb-1.0.dll. Please give it a try.

[libusb-1.0.zip](https://github.com/LibUsbDotNet/LibUsbDotNet/files/10368060/libusb-1.0.zip)

BlueFinBima, 2023-01-08:
@mcuee Thanks.  I'd already tried the version from the libusb project without success.  The one from MSYS2 also did not improve the situation (ie same exception).  Given that there appears to be a problem with the net45 version of libusbdotnet.dll calling libusb-1.0, rather than libusb-1.0.dll, I'm wondering if libusb-1.0.dll is a red-herring and there are other problems with the build of libusbdotnet.dll for net45.

mcuee, 2023-01-08:
@qmfrederik 

Please take a look at this one. Thanks.

BlueFinBima, 2023-01-08:
I'm definitely not an expert, but it certainly appears to me that at least the dll naming / resolving issue is due to there not being any resolution code for NET45 in NativeLibraryResolver.cs, so the constant LibUsbNativeLibrary is used unchanged in NativeMethods.cs, and  LibUsbNativeLibrary is `libusb-1.0`.
Unfortunately, I am unable to get my fork to build so I can't test.

MatthewPlayer, 2023-01-08:
I'm not in the office today so I cannot test today but the issue I am having is not the same as @BlueFinBima.

BlueFinBima, 2023-01-08:
@MatthewPlayer apologies for the hijack.  Not my intention.  Happy to open a new issue if @mcuee advises.

mcuee, 2023-01-08:
@BlueFinBima 

Yes please create a new issue. Thanks.

BlueFinBima, 2023-01-09:
@mcuee my issue is not reported in https://github.com/LibUsbDotNet/LibUsbDotNet/issues/183

closed 2025-04-12 by mcuee