LibUsbDotNet/LibUsbDotNet #183: win-x64 .NET Framework 4.7.2 libusb-1.0.dll from nuget package 3.0.102 alpha causes DllNotFoundException

BlueFinBima, 2023-01-09:
This problem is not restricted to alpha 3.0.102.
 .Net 5/6 builds work fine, but .Net Framework has the problem. As mentioned in another issue, renaming libusb-1.0.dll to libusb-1.0 allows it to be found, however I then get a BadImageFormatException or Memory Access exception.
 
 Specifically, I have two projects (with the same application code) both tested on Windows x64, the working one is a Console Application targeting .Net 6.0 and this works correctly with 3.0.102.alpha (both from the action and from NuGet).
The project exhibiting the problem, is  a "Console App (.NET Framework)" targeting .NET Framework 4.7.2.
### Test 1:   Built with Locally Installed NuGet Package from the action
```
Unhandled Exception: System.DllNotFoundException: Unable to load DLL 'libusb-1.0': The specified module could not be found. (Exception from HRESULT: 0x8007007E)
   at LibUsbDotNet.NativeMethods.Init(IntPtr& ctx)
   at LibUsbDotNet.LibUsb.UsbContext..ctor()
   at Examples.USBTesting.Main(String[] args) in E:\users\larmour\Source\Repos\BlueFinBima\USBTest\USBTest472\Program.cs:line 22
``` 
```
28/10/2022  13:46           173,056 libusb-1.0.dll
28/10/2022  13:47            53,248 LibUsbDotNet.dll
19/07/2017  10:01            27,992 System.Buffers.dll
06/06/2018  13:29           148,624 System.Memory.dll
19/07/2017  10:01           115,936 System.Numerics.Vectors.dll
15/05/2018  13:29            22,160 System.Runtime.CompilerServices.Unsafe.dll
08/01/2023  08:27             5,632 USBTest472.exe
05/01/2023  12:10               189 USBTest472.exe.config
08/01/2023  08:27            22,016 USBTest472.pdb
```
### Test 2:   Built with Locally Installed NuGet Package from the action and libusb-1.0.dll **renamed** to libusb-1.0
```
Unhandled Exception: System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
   at LibUsbDotNet.NativeMethods.UnrefDevice(IntPtr dev)
   at LibUsbDotNet.Device.ReleaseHandle()
   at System.Runtime.InteropServices.SafeHandle.InternalFinalize()
   at System.Runtime.InteropServices.SafeHandle.Finalize()
```
```
28/10/2022  13:46           173,056 libusb-1.0
28/10/2022  13:47            53,248 LibUsbDotNet.dll
19/07/2017  10:01            27,992 System.Buffers.dll
06/06/2018  13:29           148,624 System.Memory.dll
19/07/2017  10:01           115,936 System.Numerics.Vectors.dll
15/05/2018  13:29            22,160 System.Runtime.CompilerServices.Unsafe.dll
08/01/2023  08:27             5,632 USBTest472.exe
05/01/2023  12:10               189 USBTest472.exe.config
08/01/2023  08:27            22,016 USBTest472.pdb
``` 
### Test 3:   Built with Locally Installed NuGet Package from the action and libusb-1.0.dll from the action libusb-3345889694.zip \x64-windows\bin (timestamp 13:45)
Same results as Test 1
### Test 2:   Built with Locally Installed NuGet Package from the action and libusb-1.0.dll from the action libusb-3345889694.zip \x64-windows\bin (timestamp 13:45) **renamed** to libusb-1.0
Same results as Test 2

I'm definitely not an expert, but it certainly appears to me that at least the dll naming / resolving issue is due to there not being any resolution code for NET45 in NativeLibraryResolver.cs, so the constant `LibUsbNativeLibrary` is used unchanged in NativeMethods.cs, and  `LibUsbNativeLibrary` is `libusb-1.0`.
Unfortunately, I am unable to get my fork to build so I can't test.

BlueFinBima, 2023-01-09:
I realise that the code I'm talking about is generated, so not a permanent fix, but a "quick and dirty" test might be changing
[src/LibUsbDotNet/Generated)](https://github.com/LibUsbDotNet/LibUsbDotNet/tree/master/src/LibUsbDotNet/Generated)/NativeMethods.cs 
` public const string LibUsbNativeLibrary = "libusb-1.0";`
to
```
#if NET45
        public const string LibUsbNativeLibrary = "libusb-1.0.dll";
#else
        public const string LibUsbNativeLibrary = "libusb-1.0";
#endif
```

I can test pretty easily, or provide additional information if required.

RevanProdigalKnight, 2023-02-02:
I'm also encountering this when targeting .NET Framework 4.8.0

closed 2025-04-12 by mcuee