LibUsbDotNet/LibUsbDotNet #185: LibUsbDotNet do not handle WinUSB device using WCID

MatCDev, 2023-01-23:
I'm currently working on an USB device and a C# application. The objective is to migrate from libusbK driver (currently used) to the WinUSB driver. I've already done quite a bite of testing (including fixing a bug that prevented LibUsbDotNet from working with libusbK devices. Issue : [](https://github.com/LibUsbDotNet/LibUsbDotNet/issues/158) and PR : [](https://github.com/LibUsbDotNet/LibUsbDotNet/pull/162) ).
All of these tests were done using the Zadig to pass my device off as a WinUSB device. Everything was working perfectly.
Now the USB descriptors on the firmware side of the device have been modified. It is now natively recognized as a WinUSB device. It therefore uses the WCID. However the LibUsbDotNet library can no longer find it.
I have already done my research. The problem comes from the way to retrieve the GUID of the interfaces in the Registry. Indeed, currently, we rely on the "DeviceInterfaceGuids" properties. However, when the device is a native WinUSB device, it does not have "DeviceInterfaceGuids" properties. On the other hand, it has a "DeviceInterfaceGuid" properties that must be used. This information has already been put forward in the libusbK project (cf. [](https://github.com/mcuee/libusbk/issues/4)). LibusbK had even been modified to be able to handle this case. But since here LibUsbDotNet is not based on libusbK.dll but directly on calls to winusb.dll. The case cited above is not supported. 
I have already made the necessary changes to fix this issue in my fork of the project. I'll do a PR and let you analyze that.
The purpose of the changes made is therefore to keep the same system (via DeviceInterfaceGuids) to be able to continue to manage cases where the device is a WinUSB device via Zadig (or custom winusb inf file) + without WCID setup. 
And in the case that GetCustomDeviceProperty does not work on "DeviceInterfaceGuids", it tests to do a GetCustomDeviceProperty on the "DeviceInterfaceGuid" property.

MatCDev, 2023-01-23:
PR : [https://github.com/LibUsbDotNet/LibUsbDotNet/pull/186](https://github.com/LibUsbDotNet/LibUsbDotNet/pull/186)

mcuee, 2023-01-24:
Great. The PR looks good to me.

mcuee, 2023-01-24:
PR #186 merged.

closed 2023-01-24 by mcuee