LibUsbDotNet/LibUsbDotNet #187: Poor or no results from ShowInfo examples

MV10, 2023-01-29:
I'm using .NET 6 and Windows 10. I'm trying to use the ShowInfo examples to list my devices. I figured that would be a good, basic starting point to working with the library, but it doesn't seem to work correctly.

When I use the currently published stable package (2.2.29) and the SourceForge version of ShowInfo, I get nothing. The `Count` property on the collection returned from `UsbDevice.AllDevices` is zero. When I use the 3.x alpha package and the ShowInfo code here in the GitHub repo, I get a lot of mostly-blank output for 15 devices. The output looks like the contents of the collapsible block shown below.

<details><summary>Click to expand</summary>
<pre>

129

130

129
2

129
UR23 USB SPDIF Rx

130

130

129

130

129

130

132

129
1

Press any key to exit
</pre>
</details>

On the other hand, if I enumerate devices through WMI (for example, as shown in [this](https://stackoverflow.com/a/48390286/152997) Stackoverflow answer), I get the kind of details I'd expect (I won't paste the 5200+ rows of output, but here is one device's info in the collapsible section below). Obviously querying WMI is an entirely different approach but this is more like what I expected to see.

<details><summary>Click to expand</summary>
<pre>
DEVICE ----------------------------------
Availability:
Caption: American Power Conversion USB UPS
ClassGuid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
CompatibleID:
  USB\Class_03&SubClass_00&Prot_00
  USB\Class_03&SubClass_00
  USB\Class_03
ConfigManagerErrorCode: 0
ConfigManagerUserConfig: False
CreationClassName: Win32_PnPEntity
Description: American Power Conversion USB UPS
DeviceID: USB\VID_051D&PID_0003\3S2211X10713____
ErrorCleared:
ErrorDescription:
HardwareID:
  USB\VID_051D&PID_0003&REV_0001
  USB\VID_051D&PID_0003
InstallDate:
LastErrorCode:
Manufacturer: American Power Conversion
Name: American Power Conversion USB UPS
PNPClass: HIDClass
PNPDeviceID: USB\VID_051D&PID_0003\3S2211X10713____
PowerManagementCapabilities:
PowerManagementSupported:
Present: True
Service: HidUsb
Status: OK
StatusInfo:
SystemCreationClassName: Win32_ComputerSystem
SystemName: IG88
</pre>
</details>

Not really sure where to go from here...

mcuee, 2023-01-30:
Show.Info is broken. You may want to start with the Test_Info example.

For v2 you may want to use the more-up-to-date git v2 branch.
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2

mcuee, 2023-01-30:
Latest v2 nupkg file for your reference.
[LibUsbDotNet.2.2.46.zip](https://github.com/LibUsbDotNet/LibUsbDotNet/files/10536216/LibUsbDotNet.2.2.46.zip)

MV10, 2023-04-09:
Thanks, finally had some time to get back to this. I was already using a v2 package (2.2.29) but I switched to the 2.2.46 package you linked. Now I'm using the Show.Info example from the v2 branch and I still get no output. `UsbDevice.AllDevices` is always empty.

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/v2/stage/Examples/Show.Info/ShowInfo.cs

mcuee, 2023-04-15:
Hmm. strange. That should not happen.

Do you have any device with WinUSB driver or libusbK.sys or libusb0.sys driver?

MV10, 2023-04-15:
I don't have libusbK.sys or libusb0.sys anywhere on my drive, so I assume nothing uses them.

For WinUSB.sys, I do find that in directories which have "amd64" in them, so I'm guessing they were installed by one of the motherboard chipset updates:

[image]

mcuee, 2023-04-16:
> I don't have libusbK.sys or libusb0.sys anywhere on my drive, so I assume nothing uses them.
> 
> For WinUSB.sys, I do find that in directories which have "amd64" in them, so I'm guessing they were installed by one of the motherboard chipset updates:

No, I am not talking about the presense of WinUSB.sys. Rather are you using WinUSB.sys for any of your USB device. For libusbdotnet v2 branch, by default you need to have devices using WinUSB or libusbK.sys or libusb0.sys (libusbK.sys did not work but it is now fixed in the git version) in order to find the device.

For libusbdotnet master branch (v3 branch), it uses libusb-1.0 so you can find all the USB devices no matter the driver used. However, if you want to open the device for read/write, you will still need to use the supported driver (WinUSB, libusbK.sys or libusb0.sys).

Please refer to the following,
Ref: https://github.com/libusb/libusb/wiki/Windows#user-content-Driver_Installation

mcuee, 2023-04-16:
@sdijoseph

Just wonder if you can help fixing this in the master branch. And I think it is also a good idea to enhance this example with the code snipts from you here. Thanks.
* https://github.com/LibUsbDotNet/LibUsbDotNet/pull/192#issuecomment-1501013136

MV10, 2023-04-16:
As far as usage goes, I've found a few references in multiple places in the registry, all with the same list. 

For example: `Computer\HKEY_LOCAL_MACHINE\SYSTEM\ControlSet001\Services\WINUSB`

The `Owners` key has these listed:

```
oem86.inf
wpdmtp.inf
oem76.inf
oem52.inf
winusb.inf
transfercable.inf
```

Based on the INF content, some of these are devices I have used in the past (Android ADB drivers, SiliconLabs drivers used for Arduino and STM32 programming, a Belkin USB transfer cable) but nothing that would be connected all the time (or when I was testing).

I'm currently just trying to enumerate all USB devices as a starting point (and maybe this library isn't the right way?), but the device I'm ultimately interested in is not an HID device, it is my APC UPS backup power supply. I was hoping to write something to monitor the UPS for power events and send myself notifications. Knowing that, would "installing a USB driver" (WinUSB) associated with that device interfere with the built-in Windows support for the UPS?

mcuee, 2023-04-17:
@MV10 

PR #207 has been merged and now you should be able to use `ShowInfo` example from the master branch as well.

As for your APC USP Backup power supply, if it is using WinUSB driver (you can check from Device Manager), then yes `ShowInfo` example should be able to print out the device information. However, you need to know the communication protocol in order to carry out meaningful communication with the device.

If the device is not using WinUSB driver, then you have to use Zadig to use WinUSB driver to replace the vendor driver. But then the vendor application will not work.

I suspect that libusb/libusbdotnet is not the right library for you.

But since this is not a support forum, you may have to ask elsewhere, like libusb-devel mailing list.

closed 2023-04-17 by mcuee