LibUsbDotNet/LibUsbDotNet #188: UsbDevice.GetStringDescriptor throws Input/Output Error on Linux

ClusterM, 2023-02-01:
Hello.

Library version: 3.0.102-alpha

My code:
```C#
var allDevices = context.List();
var upses = allDevices.Where(d => d.ProductId == 0 && d.VendorId == 1);
if (!upses.Any()) throw new IOException("Can't find UPS");
var ups = upses.First();
ups.Open();
var s = ups.GetStringDescriptor(3); // Stats
```
It works fine on Windows, but throws `LibUsbDotNet.LibUsb.UsbException`: "Input/Output Error" on Linux.
Stack trace:
```
 at LibUsbDotNet.LibUsb.ErrorExtensions.ThrowOnError(Error error)
   at LibUsbDotNet.LibUsb.UsbDevice.GetStringDescriptor(Byte descriptorIndex, Boolean failSilently)
```

mcuee, 2023-02-02:
You may want to check if libusb examples work with your devices first before trying libusbdotnet as you are using an embedded Linux distribution.

For example, you can try the `xusb` example.
https://github.com/libusb/libusb/tree/master/examples

mcuee, 2023-02-02:
Another thing, the v2 branch may be more stable and you may want to try it instead.
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2

ClusterM, 2023-02-03:
> You may want to check if libusb examples work with your devices first before trying libusbdotnet as you are using an embedded Linux distribution.
> 
> For example, you can try the `xusb` example. https://github.com/libusb/libusb/tree/master/examples

It's not working too. It can't even read manufacturer string. Seems like it's very Chinese device :)

closed 2023-02-03 by ClusterM