LibUsbDotNet/LibUsbDotNet #189: epwriter.write win32error

ChineseK, 2023-02-13:
public bool openDevice(int index)
        {
            mDev = allDevices[listUsb[index]];

            UsbDeviceFinder usbFinder = new UsbDeviceFinder(22918, 37126);
            var usbDevice = UsbDevice.OpenUsbDevice(usbFinder);

            IUsbDevice wholeUSBDevice = usbDevice as IUsbDevice;
            if (wholeUSBDevice != null)
            {
                // Select config #1
                wholeUSBDevice.SetConfiguration(1);

                // Claim interface #0.
                wholeUSBDevice.ClaimInterface(0);
            }

            epReader = usbDevice.OpenEndpointReader(ReadEndpointID.Ep01);
            epWriter = usbDevice.OpenEndpointWriter(WriteEndpointID.Ep01);
            epReader.DataReceived += EpReader_DataReceived;
            return true;
        }
        
 private void Write()
{
   var ec =epWriter.Write(array, 2000, out length);
  // ec = win32error
}       
I can open device  but write error  ,return code win32error

mcuee, 2023-02-13:
What is this device? Do you know the underlying communication protocol of this device? You can not anyhow write to the device without knowing the communication protocol.

ChineseK, 2023-02-13:
The communication protocol is given by the manufacturer
 public byte XMT_COMMAND_Assist_ReadFlag(int usbDeviceNum, byte address, byte Command_B3, byte Command_B4, byte Channel_Num)
        {
            byte b = 7;
            byte[] array = new byte[b];
            array[0] = 170;
            array[1] = address;
            array[2] = b;
            array[3] = Command_B3;
            array[4] = Command_B4;
            array[5] = Channel_Num;
            byte b2 = 0;
            for (int i = 0; i < b - 1; i++)
            {
                b2 = (byte)(b2 ^ array[i]);
            }

            array[b - 1] = b2;
            mEpWriter.Write(array, 2000);
            mEpReader.Read(tmp_command, 50);
            b2 = 0;
            for (int i = 0; i < 7; i++)
            {
                b2 = (byte)(b2 ^ tmp_command[i]);
            }

            byte result = 119;
            if (b2 == tmp_command[7])
            {
                return tmp_command[6];
            }

            return result;
        }
        Why do you think it's a communication protocol problem  = Win32 error
        This gives me a  idea about this problem

mcuee, 2023-02-13:
>    Why do you think it's a communication protocol problem  = Win32 error
>    This gives me a  idea about this problem

Typically timeout means that the device does not understand the command sent to the device. Therefore I think your codes are not following the communication protocol.

ChineseK, 2023-02-13:
Thanks , i will try to get more error information later

ChineseK, 2023-02-13:
Win32Error:PipeTransferSubmit Ep 0x01
87:The parameter is incorrect.

ForceLibUsbWinBack=1
Someone asked me to set it this way

but I don't understand whether it is useful

ChineseK, 2023-02-13:
Win32Error:PipeTransferSubmit Ep 0x01
87:The parameter is incorrect.

ChineseK, 2023-02-13:
maybe Ep 0x01 or  communication protocol byte[] is wrong?

mcuee, 2023-02-15:
For this issue, we can not help you much as you have to understand the communication protocol to talk to the device.

mcuee, 2023-02-15:
This github repo is only for reporting libusbdotnet issues and not for technical support.

closed 2023-02-15 by mcuee