LibUsbDotNet/LibUsbDotNet #190: 编译V2分支无法获取USB 存储设备信息

FreedomPointer, 2023-02-13:
作者，你好！
我编译V2分支后，调用接口仍然获取不到USB 存储盘信息(如”序列号“这样的信息)
还有，必须安装这个软件吗？
但好像只支持windows，如何跨平台呢？
[image]

ChineseK, 2023-02-15:
必须安装这个软件

mcuee, 2023-02-15:
@FreedomPointer 

Please read the following libusb Wiki pages. 
https://github.com/libusb/libusb/wiki/Windows#user-content-Driver_Installation
https://github.com/libusb/libusb/wiki/FAQ

And please DO NOT use libusb/libusbdotnet with USB Mass Storage Device. It is not the right library for your task. 

You will not be able to use it under macOS as there is no way to detach the system driver. 

You have to replace the system USB mass storage driver with a supported driver under Windows.

mcuee, 2023-02-15:
This is not a libusbdotnet issue and I will close this one.

closed 2023-02-15 by mcuee

FreedomPointer, 2023-02-15:
非常感谢！
@mcuee 
@ChineseK