LibUsbDotNet/LibUsbDotNet #191: Incorrect fieldnames in LangStringDescriptor

Juliusqaz123, 2023-02-14:
I am currently using v3.0.102-alpha version of LibUsbDotnet. I noticed that `bool GetString(out string stringData, short langId, byte stringIndex);` always crashes with following exception : 

```
Device Firmware Upgrade failed with exception: System.TypeInitializationException: The type initializer for 'LibUsbDotNet.Descriptors.LangStringDescriptor' threw an exception.
 ---> System.ArgumentException: Field passed in is not a marshaled member of the type 'LibUsbDotNet.Descriptors.UsbDescriptor'. (Parameter 'fieldName')
```

After viewing source i noticed that these 2 fields are causing problem:
`  private static readonly int OfsDescriptorType = Marshal.OffsetOf(typeof(UsbDescriptor), "descriptorType").ToInt32();
   private static readonly int OfsLength = Marshal.OffsetOf(typeof(UsbDescriptor), "length").ToInt32();`
   
 In UsbDescriptor class fieldnames are returned starting with capital letters like this:
 
 ` public override string ToString()
        {
            object[] values = { this.length, this.descriptorType };
            string[] names = { "Length", "DescriptorType" };

            return Helper.ToString(string.Empty, names, ToStringParamValueSeperator, values, ToStringFieldSeperator);
        }`
        
   So LangStringDescriptor always crashes because it looks for descriptorType and length, not the DescriptorType and Length.
   I also noticed that in 2.29 version that LangStringDescriptor  has correct fieldnames with capital letters. Am i missing something or is it bug?

mcuee, 2023-02-14:
@Juliusqaz123 
Looks like a bug to me. Please help to create a PR. Thanks.

@qmfrederik
Please confirm that this is a problem as well. Thanks.

Juliusqaz123, 2023-02-15:
@mcuee I ain't exactly sure from which branch can i branch off, cause there is no branch for v3 as far as can i see. I saw this bug only from downloading nuget and viewing source.

mcuee, 2023-02-17:
> @mcuee I ain't exactly sure from which branch can i branch off, cause there is no branch for v3 as far as can i see. I saw this bug only from downloading nuget and viewing source.

V3 means the master branch, the main development branch by the current developer @qmfrederik 
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/master

V2 branch is the continuation of the original libusbdotnet implementation by the original developer Travis Robinson.
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/v2

Juliusqaz123, 2023-02-17:
@mcuee  Well LangStringDescriptor differs between the master branch and the one downloaded as nuget (3.0.102-alpha). So I ain't quite sure how to create pull request

mcuee, 2023-02-17:
I see. No problem. 

Let's wait for @qmfrederik to check if this is still an issue or not in the master branch.

mcuee, 2023-04-15:
https://nuget.info/packages/LibUsbDotNet/3.0.102-alpha
3.0.102-alpha should be the same as https://github.com/LibUsbDotNet/LibUsbDotNet/commit/27e68979a1f4036783fb0eca759c6ed1492e76cb

mcuee, 2023-04-15:
I think this is not valid.
https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/LibUsbDotNet/Descriptors/UsbDescriptor.cs

```
        public override string ToString()
        {
            object[] values = { this.length, this.descriptorType };
            string[] names = { "Length", "DescriptorType" };

            return Helper.ToString(string.Empty, names, ToStringParamValueSeperator, values, ToStringFieldSeperator);
        }

```

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/LibUsbDotNet/Descriptors/LangStringDescriptor.cs

```
        #region FIELD_OFFSETS

        private static readonly int OfsDescriptorType = Marshal.OffsetOf(typeof(UsbDescriptor), "DescriptorType").ToInt32();
        private static readonly int OfsLength = Marshal.OffsetOf(typeof(UsbDescriptor), "Length").ToInt32();

        #endregion
```

mcuee, 2023-04-15:
@Juliusqaz123 

Please post an example to show the problem. I will reopen the issue if I can reproduce the issue.

Juliusqaz123, 2023-05-04:
here's inner exception trace:

System.TypeInitializationException: The type initializer for 'LibUsbDotNet.Descriptors.LangStringDescriptor' threw an exception.
 ---> System.ArgumentException: Field passed in is not a marshaled member of the type 'LibUsbDotNet.Descriptors.UsbDescriptor'. (Parameter 'fieldName')
   at System.Runtime.InteropServices.Marshal.OffsetOf(Type t, String fieldName)
   at LibUsbDotNet.Descriptors.LangStringDescriptor..cctor()
   --- End of inner exception stack trace ---
   at LibUsbDotNet.Descriptors.LangStringDescriptor.get_Length()
   at LibUsbDotNet.LibUsb.UsbDevice.GetString(String& stringData, Int16 langId, Byte stringIndex)
   
  here's a little examplary console app i wrote which shows the problem:
   
   `using System.Runtime.InteropServices;
using System.Text;

var result = LangStringDescriptor.OfsDescriptorType;
Console.WriteLine(result);

public class LangStringDescriptor
{
    public static readonly int OfsDescriptorType = Marshal.OffsetOf(typeof(UsbDescriptor), "DescriptorType").ToInt32();
}

[StructLayout(LayoutKind.Sequential)]
public abstract class UsbDescriptor
{
    /// <summary>
    /// String value used to seperate the name/value pairs for all ToString overloads of the descriptor classes.
    /// </summary>
    public const string ToStringParamValueSeperator = ":";

    /// <summary>
    /// String value used to seperate the name/value groups for all ToString overloads of the descriptor classes.
    /// </summary>
    public const string ToStringFieldSeperator = "\r\n";

    /// <summary>
    /// Total size of this structure in bytes.
    /// </summary>
    public static readonly int Size = Marshal.SizeOf(typeof(UsbDescriptor));

    /// <summary>
    /// Length of structure reported by the associated usb device.
    /// </summary>
    private byte length;

    /// <summary>
    /// Type of structure reported by the associated usb device.
    /// </summary>
    private DescriptorType descriptorType;

    /// <inheritdoc/>
    public override string ToString()
    {
        object[] values = {this.length, this.descriptorType};
        string[] names = {"Length", "DescriptorType"};

        return Helper.ToString(string.Empty, names, ToStringParamValueSeperator, values, ToStringFieldSeperator);
    }
}

public class DescriptorType
{
    
}

public static class Helper
{
    /// <summary>
    /// Builds a delimited string of names and values.
    /// </summary>
    /// <param name="sep0">Inserted and the begining of the entity.</param>
    /// <param name="names">The list of names for the object values.</param>
    /// <param name="sep1">Inserted between the name and value.</param>
    /// <param name="values">The values for the names.</param>
    /// <param name="sep2">Inserted and the end of the entity.</param>
    /// <returns>The formatted string.</returns>
    public static string ToString(string sep0, string[] names, string sep1, object[] values, string sep2)
    {
        StringBuilder sb = new StringBuilder();

        for (int i = 0; i < names.Length; i++)
        {
            sb.Append(sep0 + names[i] + sep1 + values[i] + sep2);
        }

        return sb.ToString();
    }
}`

Juliusqaz123, 2023-05-04:
Here's inner stacktrace i from exception i received:

System.TypeInitializationException: The type initializer for 'LibUsbDotNet.Descriptors.LangStringDescriptor' threw an exception.
 ---> System.ArgumentException: Field passed in is not a marshaled member of the type 'LibUsbDotNet.Descriptors.UsbDescriptor'. (Parameter 'fieldName')
   at System.Runtime.InteropServices.Marshal.OffsetOf(Type t, String fieldName)
   at LibUsbDotNet.Descriptors.LangStringDescriptor..cctor()
   --- End of inner exception stack trace ---
   at LibUsbDotNet.Descriptors.LangStringDescriptor.get_Length()
   at LibUsbDotNet.LibUsb.UsbDevice.GetString(String& stringData, Int16 langId, Byte stringIndex)
   
 And here's the console application i wrote which shows the problem:
 
 `using System.Runtime.InteropServices;
using System.Text;

var result = LangStringDescriptor.OfsDescriptorType;
Console.WriteLine(result);`

`public class LangStringDescriptor
{
    public static readonly int OfsDescriptorType = Marshal.OffsetOf(typeof(UsbDescriptor), "DescriptorType").ToInt32();
}`

`[StructLayout(LayoutKind.Sequential)]
public abstract class UsbDescriptor
{
    /// <summary>
    /// String value used to seperate the name/value pairs for all ToString overloads of the descriptor classes.
    /// </summary>
    public const string ToStringParamValueSeperator = ":";

    /// <summary>
    /// String value used to seperate the name/value groups for all ToString overloads of the descriptor classes.
    /// </summary>
    public const string ToStringFieldSeperator = "\r\n";

    /// <summary>
    /// Total size of this structure in bytes.
    /// </summary>
    public static readonly int Size = Marshal.SizeOf(typeof(UsbDescriptor));

    /// <summary>
    /// Length of structure reported by the associated usb device.
    /// </summary>
    private byte length;

    /// <summary>
    /// Type of structure reported by the associated usb device.
    /// </summary>
    private DescriptorType descriptorType;

    /// <inheritdoc/>
    public override string ToString()
    {
        object[] values = {this.length, this.descriptorType};
        string[] names = {"Length", "DescriptorType"};

        return Helper.ToString(string.Empty, names, ToStringParamValueSeperator, values, ToStringFieldSeperator);
    }
}`

`public class DescriptorType
{
    
}`

`public static class Helper
{
    /// <summary>
    /// Builds a delimited string of names and values.
    /// </summary>
    /// <param name="sep0">Inserted and the begining of the entity.</param>
    /// <param name="names">The list of names for the object values.</param>
    /// <param name="sep1">Inserted between the name and value.</param>
    /// <param name="values">The values for the names.</param>
    /// <param name="sep2">Inserted and the end of the entity.</param>
    /// <returns>The formatted string.</returns>
    public static string ToString(string sep0, string[] names, string sep1, object[] values, string sep2)
    {
        StringBuilder sb = new StringBuilder();

        for (int i = 0; i < names.Length; i++)
        {
            sb.Append(sep0 + names[i] + sep1 + values[i] + sep2);
        }

        return sb.ToString();
    }
}`

The code snippets from the master branch. @mcuee

Juliusqaz123, 2023-05-04:
Here's the inner trace from exception I received:

System.TypeInitializationException: The type initializer for 'LibUsbDotNet.Descriptors.LangStringDescriptor' threw an exception.
 ---> System.ArgumentException: Field passed in is not a marshaled member of the type 'LibUsbDotNet.Descriptors.UsbDescriptor'. (Parameter 'fieldName')
   at System.Runtime.InteropServices.Marshal.OffsetOf(Type t, String fieldName)
   at LibUsbDotNet.Descriptors.LangStringDescriptor..cctor()

Here's the example console app which shows the problem created from code snippets (master branch). 
   
```   
using System.Runtime.InteropServices;
using System.Text;

var result = LangStringDescriptor.OfsDescriptorType;
Console.WriteLine(result);

public class LangStringDescriptor
{
    public static readonly int OfsDescriptorType = Marshal.OffsetOf(typeof(UsbDescriptor), "DescriptorType").ToInt32();
}

[StructLayout(LayoutKind.Sequential)]
public abstract class UsbDescriptor
{
    /// <summary>
    /// String value used to seperate the name/value pairs for all ToString overloads of the descriptor classes.
    /// </summary>
    public const string ToStringParamValueSeperator = ":";

    /// <summary>
    /// String value used to seperate the name/value groups for all ToString overloads of the descriptor classes.
    /// </summary>
    public const string ToStringFieldSeperator = "\r\n";

    /// <summary>
    /// Total size of this structure in bytes.
    /// </summary>
    public static readonly int Size = Marshal.SizeOf(typeof(UsbDescriptor));

    /// <summary>
    /// Length of structure reported by the associated usb device.
    /// </summary>
    private byte length;

    /// <summary>
    /// Type of structure reported by the associated usb device.
    /// </summary>
    private DescriptorType descriptorType;

    /// <inheritdoc/>
    public override string ToString()
    {
        object[] values = {this.length, this.descriptorType};
        string[] names = {"Length", "DescriptorType"};

        return Helper.ToString(string.Empty, names, ToStringParamValueSeperator, values, ToStringFieldSeperator);
    }
}

public class DescriptorType
{
    
}

public static class Helper
{
    /// <summary>
    /// Builds a delimited string of names and values.
    /// </summary>
    /// <param name="sep0">Inserted and the begining of the entity.</param>
    /// <param name="names">The list of names for the object values.</param>
    /// <param name="sep1">Inserted between the name and value.</param>
    /// <param name="values">The values for the names.</param>
    /// <param name="sep2">Inserted and the end of the entity.</param>
    /// <returns>The formatted string.</returns>
    public static string ToString(string sep0, string[] names, string sep1, object[] values, string sep2)
    {
        StringBuilder sb = new StringBuilder();

        for (int i = 0; i < names.Length; i++)
        {
            sb.Append(sep0 + names[i] + sep1 + values[i] + sep2);
        }

        return sb.ToString();
    }
}
```
hope it helps
@mcuee

mcuee, 2023-05-06:
@Juliusqaz123 

Can you give the latest git master a try? You can get the nuget package here (3.0.122-alpha).
https://github.com/LibUsbDotNet/LibUsbDotNet/suites/12408152177/artifacts/659872788

mcuee, 2023-09-21:
@Juliusqaz123 

Just wondering if you have any updates? Thanks. Please build libusbdotnet from latest git source.

mcuee, 2024-05-05:
No update for a while, I will close this issue for now. Please reopen if the issue is still there.

closed 2024-05-05 by mcuee