LibUsbDotNet/LibUsbDotNet #195: Device always returning null

vidhya-vishwa, 2023-04-04:
I create a .net code with v2 example code but the device is always returning null for me even i open device with vendor ID ,Product ID or with revision or via myusbfinder the device is null
`   foreach (KeyValuePair<string, object> kvp in usbRegistry.DeviceProperties)
                {
                   if(kvp.Key.Contains("Hardware"))
                   {
                       string[] arr = Array.ConvertAll((object[])kvp.Value, Convert.ToString);
                       string[] hdlist = arr[0].Split("&");
                      
                       foreach (var item in hdlist)
                       {
                                    if(item.Contains("Vid"))
                                        vid=item.Replace("Vid_","");
                                    if(item.Contains("Pid"))
                                        pid=item.Replace("Pid_","");
                                    if(item.Contains("Rev"))
                                        rev=item.Replace("Rev__","");
                           Console.WriteLine(item.ToString());
                       }
                       
                   }
                }
                    Console.WriteLine("vid"+vid+" pid"+pid);
                    // int vid_int = Int32.Parse(vid);
                    // int pid_int = Int32.Parse(pid);
                    UsbDeviceFinder MyUsbFinder = new UsbDeviceFinder(rev);
                    MyUsbDevice = UsbDevice.OpenUsbDevice(MyUsbFinder);
                
                // If the device is open and ready
                if (MyUsbDevice == null) throw new Exception("Device Not Found.");

`

MyUsbDevice always returns null

mcuee, 2023-04-23:
Maybe you want to use Show.Info example from the v2 branch as a start. It works fine under Linux.

```
mcuee@UbuntuSwift3:~/build/libusb/libusbdotnet_v2$ ./stage/Examples/Show.Info/bin/Release/net6.0/Show.Info 
Length:18
DescriptorType:Device
BcdUsb:0x0200
Class:VendorSpec
SubClass:0x00
Protocol:0x00
MaxPacketSize0:64
VendorID:0x1C7A
ProductID:0x0575
BcdDevice:0x1072
ManufacturerStringIndex:1
ProductStringIndex:2
SerialStringIndex:3
ConfigurationCount:1
ManufacturerString:EgisTec
ProductString:EgisTec EH575
SerialString:077E2F9A

Length:9
DescriptorType:Configuration
TotalLength:46
InterfaceCount:1
ConfigID:1
StringIndex:0
Attributes:0xA0
MaxPower:50
ConfigString:

Length:9
DescriptorType:Interface
InterfaceID:0
AlternateID:0
EndpointCount:4
Class:VendorSpec
SubClass:0xFF
Protocol:0x00
StringIndex:0
InterfaceString:

Length:7
DescriptorType:Endpoint
EndpointID:0x01
Attributes:0x02
MaxPacketSize:512
Interval:0
Refresh:0
SynchAddress:0x00

Length:7
DescriptorType:Endpoint
EndpointID:0x82
Attributes:0x02
MaxPacketSize:512
Interval:0
Refresh:0
SynchAddress:0x00

Length:7
DescriptorType:Endpoint
EndpointID:0x83
Attributes:0x03
MaxPacketSize:16
Interval:8
Refresh:0
SynchAddress:0x00

Length:7
DescriptorType:Endpoint
EndpointID:0x84
Attributes:0x03
MaxPacketSize:16
Interval:8
Refresh:0
SynchAddress:0x00

Length:18
DescriptorType:Device
BcdUsb:0x0200
Class:PerInterface
SubClass:0x00
Protocol:0x00
MaxPacketSize0:8
VendorID:0x04D8
ProductID:0xFA2E
BcdDevice:0x0001
ManufacturerStringIndex:1
ProductStringIndex:2
SerialStringIndex:3
ConfigurationCount:1
ManufacturerString:Travis Robinson
ProductString:Benchmark Device
SerialString:LUSBW1

Length:9
DescriptorType:Configuration
TotalLength:32
InterfaceCount:1
ConfigID:1
StringIndex:0
Attributes:0xC0
MaxPower:50
ConfigString:

Length:9
DescriptorType:Interface
InterfaceID:0
AlternateID:0
EndpointCount:2
Class:PerInterface
SubClass:0x00
Protocol:0x00
StringIndex:0
InterfaceString:

Length:7
DescriptorType:Endpoint
EndpointID:0x01
Attributes:0x02
MaxPacketSize:32
Interval:0
Refresh:0
SynchAddress:0x00

Length:7
DescriptorType:Endpoint
EndpointID:0x81
Attributes:0x02
MaxPacketSize:32
Interval:0
Refresh:0
SynchAddress:0x00

```

mcuee, 2023-04-23:
You can also switch to the master branch and use Show.Info example as well (from git head).

closed 2023-04-23 by mcuee