LibUsbDotNet/LibUsbDotNet #197: Test_Info is broken in master (v3) branch

mcuee, 2023-04-09:
As of now, Test_Info is broken in master (v3) branch. v2 branch is okay.

Detailed of the exception under Windows.

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.NullReferenceException: Object reference not set to an instance of an object.
   at Test_Info.fTestInfo.addDevice(IUsbDevice deviceReg, String display) in C:\work\libusb\libusbdotnet\LibUsbDotNet\src\Test_Info\fTestInfo.cs:line 99
   at Test_Info.fTestInfo.cboDevices_SelectedIndexChanged(Object sender, EventArgs e) in C:\work\libusb\libusbdotnet\LibUsbDotNet\src\Test_Info\fTestInfo.cs:line 138
   at System.Windows.Forms.ComboBox.OnSelectedIndexChanged(EventArgs e)
   at System.Windows.Forms.ComboBox.WmReflectCommand(Message& m)
   at System.Windows.Forms.ComboBox.WndProc(Message& m)
   at System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, WM msg, IntPtr wparam, IntPtr lparam)

************** Loaded Assemblies **************
System.Private.CoreLib
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Private.CoreLib.dll
----------------------------------------
Test_Info
    Assembly Version: 1.0.0.0
    Win32 Version: 1.0.0.0
    CodeBase: file:///C:/work/libusb/libusbdotnet/LibUsbDotNet/src/Test_Info/bin/Debug/net6.0-windows/Test_Info.dll
----------------------------------------
System.Runtime
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Runtime.dll
----------------------------------------
System.Windows.Forms
    Assembly Version: 6.0.2.0
    Win32 Version: 6.0.1523.11702
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.15/System.Windows.Forms.dll
----------------------------------------
System.Windows.Forms.Primitives
    Assembly Version: 6.0.2.0
    Win32 Version: 6.0.1523.11702
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.15/System.Windows.Forms.Primitives.dll
----------------------------------------
System.ComponentModel.Primitives
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.ComponentModel.Primitives.dll
----------------------------------------
System.Runtime.InteropServices
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Runtime.InteropServices.dll
----------------------------------------
System.Drawing.Primitives
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Drawing.Primitives.dll
----------------------------------------
System.Collections.Specialized
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Collections.Specialized.dll
----------------------------------------
System.Drawing.Common
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.WindowsDesktop.App/6.0.15/System.Drawing.Common.dll
----------------------------------------
System.Threading
[... 118 lines cut ...]
----------------------------------------
System.Private.Uri
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Private.Uri.dll
----------------------------------------
Microsoft.Win32.Registry
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/Microsoft.Win32.Registry.dll
----------------------------------------
System.Runtime.CompilerServices.Unsafe
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Runtime.CompilerServices.Unsafe.dll
----------------------------------------
System.Runtime.Loader
    Assembly Version: 6.0.0.0
    Win32 Version: 6.0.1523.11507
    CodeBase: file:///C:/Program%20Files/dotnet/shared/Microsoft.NETCore.App/6.0.15/System.Runtime.Loader.dll
----------------------------------------

************** JIT Debugging **************

```

sdijoseph, 2023-04-16:
I think, when I get time, I'll make a replacement for this example that uses [Avalonia](https://avaloniaui.net/). I think a cross-platform library should have a cross-platform example.

mcuee, 2023-05-19:
@sdijoseph 

Just wondering if you are planning to work on this issue. Thanks.

sdijoseph, 2023-05-19:
> @sdijoseph
> 
> Just wondering if you are planning to work on this issue. Thanks.

Yes I will get it done at some point in the next 1-2 months, I've just been a bit busy with work lately.

mcuee, 2023-05-19:
> > @sdijoseph
> > Just wondering if you are planning to work on this issue. Thanks.
> 
> Yes I will get it done at some point in the next 1-2 months, I've just been a bit busy with work lately.

No problem. This is not so important. Take your time as work is always more important. :-)

PR #268 (2025-09-25): Test info replacement
Starting the replacement of the TestInfo app from WinForms to Avalonia. Now it will run on Windows, Linux, and Mac. It is still a WIP, as right now it only displays the UsbDeviceInfo of each device. I will try to add more features over time such as:

1. On platforms that support it, auto-updating the device list via the hotplug feature.
2. Show configuration and interface information.
3. Running benchmarking samples.

Closes #197
closed 2025-09-25 by mcuee