LibUsbDotNet/LibUsbDotNet #209: Documentation for master branch

mcuee, 2023-04-16:
Right now the documentation is actually for the v2 branch. It is good to update the documentation for the master branch.

mcuee, 2023-04-16:
As mentioned in #210, I actually do not know how to build the documentation.
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/210
https://github.com/LibUsbDotNet/LibUsbDotNet/tree/master/src/Docs

mcuee, 2023-04-16:
Looks like Travis was using Sandcastle Help File Builder. But the version may be pretty old and I can not build the doc under VS2022 with the latest release of SHFB (v2023.3.4.0).
https://github.com/EWSoftware/SHFB

For the master branch, I am not so sure what is the tool to author and build the documentation.

sdijoseph, 2023-04-16:
I just tried building a local GitHub Pages site for documentation using [DocFX](https://dotnet.github.io/docfx/index.html), and it seemed to work pretty well. I think we can add more thorough XML comments across the project, and then move on to publishing the generated docs to a site.

mcuee, 2023-05-07:
@qmfrederik 

I have merged #211 into git master. However, I do not have the permission to finish the task to deploy. Please help here. Thanks.

sdijoseph, 2023-05-19:
@qmfrederik Just to specify, the steps to deploy the doc site are [here](https://docs.github.com/en/pages/quickstart#creating-your-website) (from 3 on).

Japollack, 2025-08-12:
Can you write a short comparison based on the v3 vs. v2 features? Heads-up warnings if there is any major change that breaks code, etc.