LibUsbDotNet/LibUsbDotNet #217: TryOpen() works on STM32 device via Mac, but same call fails on Windows...

CartBlanche, 2023-04-25:
I have the following code, where `ourDevices` is a list of STM32 devices in bootloader mode.

               ```
                foreach (UsbDevice device in ourDevices)
                {
                     if (device != null && device.TryOpen())
                    {
                      // Add device to bootloader flashing list...
                    }
                }
```

Running this code on MacOS (v13.3.1, Apple M2), the `TryOpen()` call is successful and I can then get the required device.Info.* information before flashing that device. Running exactly that code (it's a console/command line app for mass flashing of attached devices) on Windows 11 Pro (v21H2 - ARM), the `TryOpen()` call never succeeds, so I cannot then flash the devices from Windows.
Is there other incantation/pre-requisite required for this to work or is this a known issue?

mcuee, 2023-04-25:
In order for libusbdotnet (or libusb) to work under Windows, your device needs to use supported driver -- WinUSB (recommended), libusbK.sys and libusb0.sys.
https://github.com/libusb/libusb/wiki/Windows#driver-installation

CartBlanche, 2023-04-26:
@mcuee I just tried the latest Zadiag and the driver installation on ARM (even though they say they now support ARM) fails.
[image]

I'm new to IoT etc development, so appologies if this is a stupid question, but I need the STM32 device to be detectable on Windows, Linux and MacOS as our IDE extensions (VS for Mac, VS for Windows and VSCode) target those platform. So can I change the STM32 FW,  to support Micosoft OS Descriptor for USB Devices as well as still support Linux and Mac?

mcuee, 2023-04-26:
For Zadig issues, please report to libwdi, thanks. Which Windows on ARM64 are you using?

As for Micosoft OS Descriptor for USB Devices, it will not affect your devices under Linux and Mac.

mcuee, 2023-04-26:
As this is not a support forum, I will close the issue here. If you want further help, you may want to use libusb-devel mailing list. You need to subscribe to the mailing list in order to post.

closed 2023-04-26 by mcuee

mcuee, 2023-04-26:
@CartBlanche 

For libwdi issue, please post the details here. Thanks.
* https://github.com/pbatard/libwdi/issues/155

CartBlanche, 2023-04-26:
OK will do.

CartBlanche, 2023-04-28:
@mcuee I've now run that same code on an x64 Windows 10 Pro and it still fails on the `device.Info.SerialNumber` property call, after the `TryOpen()` looks like it succeeds. The latest WinUSB driver is installed on this machine.
This is the new error. 
```
System.AccessViolationException: 'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.'
```

I'm using v3.0.102-alpha, if that makes a difference?

Should I raise a separate issue for this? Am I doing something wrong?

mcuee, 2023-04-28:
Please post another issue. Thanks. Please provide the details of what you do and more details of the codes.