LibUsbDotNet/LibUsbDotNet #218: 3.0.102-alpha other memory is corrupt

OaimIsGnay, 2023-04-27:
3.0.102-alpha

[image]

[image]

mcuee, 2023-04-27:
What is the device and OS?

If this is an HID device, please do not use libusbdotnet or libusb under macOS since it will not work. You should use HIDAPI instead.

OaimIsGnay, 2023-04-27:
> 

windows,  Console error will be reported after running for a while.

OaimIsGnay, 2023-04-27:
> What is the device and OS?
> 
> If this is an HID device, please do not use libusbdotnet or libusb under macOS since it will not work. You should use HIDAPI instead.
> 
> Reference: https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices

I have resolved the issue and cannot initialize multiple times.

closed 2023-04-27 by OaimIsGnay

zhangxinling2, 2023-07-27:
> > 什么是设备和操作系统？
> > 如果这是HID设备，请不要在macOS下使用libusbdotnet或libusb，因为它将无法正常工作。您应该改用 HIDAPI。
> > 参考： https://github.com/libusb/libusb/wiki/FAQ#does-libusb-support-usb-hid-devices
> 
> 我已经解决了这个问题，无法多次初始化。

hello,how do you  resolve this problem