LibUsbDotNet/LibUsbDotNet #219: [Windows]

CartBlanche, 2023-05-05:
I have the following code, where `ourDevices` is a list of STM32 devices in bootloader/DFU mode.

```
                foreach (UsbDevice device in ourDevices)
                {
                     if (device != null && device.TryOpen())
                    {
                      // Add device to bootloader flashing list...
                    }
                }
```

Running this code on MacOS (v13.3.1, Apple M2), the TryOpen() call is successful and I can then get the required device.Info.* information before flashing that device. Running exactly that code (it's a console/command line app for mass flashing of attached devices) on an x64 Windows 10 Pro (not a VM this time, real hardware), it fails on the `device.Info.SerialNumber` property call, after the `TryOpen()` looks like it succeeds. The latest WinUSB driver is installed on this x64 machine.

This is the error. 
```
System.AccessViolationException: 'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.'
```

I'm using `v3.0.102-alpha`, if that makes a difference?

mcuee, 2023-05-06:
@CartBlanche 

Can you give the master a try? You can get the nuget package here.
https://github.com/LibUsbDotNet/LibUsbDotNet/suites/12408152177/artifacts/659872788

CartBlanche, 2023-05-09:
@mcuee Thanks. I'll give this a try today.

CartBlanche, 2023-05-15:
@mcuee Unfortunately I still get the AccessViolation...
```
Fatal error. System.AccessViolationException: Attempted to read or write protected memory. This is often an indication that other memory is corrupt.
Repeat 2 times:
--------------------------------
   at LibUsbDotNet.NativeMethods.GetStringDescriptorAscii(LibUsbDotNet.DeviceHandle, Byte, Byte*, Int32)
--------------------------------
   at LibUsbDotNet.LibUsb.UsbDevice.GetStringDescriptor(Byte, Boolean)
   at LibUsbDotNet.Info.UsbDeviceInfo.FromUsbDeviceDescriptor(LibUsbDotNet.LibUsb.IUsbDevice, LibUsbDotNet.DeviceDescriptor)
   at LibUsbDotNet.LibUsb.UsbDevice.get_Descriptor()
```

Is there anything else you need?

mcuee, 2023-05-15:
Thanks for the report. As you mentioned, this could be a timing issue.

I am not so sure if you can post a short reproduce code so that I can try it as well. I do have some STM32 boards if your codes need to deal with DFU.

mcuee, 2023-05-17:
@CartBlanche 

BTW, what is the version of libusb are you using? What is the driver used? Please use WinUSB driver and not libusb0.sys as there may be issues for libusb0.sys (especially the older 1.2.6.0 version).

CartBlanche, 2023-05-23:
@mcuee I was away at a conference last week, so only got back home yesterday. I think I'm going to checkout the repo locally and build it, reference it and see why it fails locally. If I find a fix, I'll send a PR for your perusal. :)

CartBlanche, 2023-07-12:
@mcuee I'm back into looking at this issue as a customer has now hit the same issue on Windows 10. Same code path works fine on Windows 11.

CartBlanche, 2023-07-13:
@mcuee, @sdijoseph 
The crash on Windows 10 happens in here (code was merged into the code base about 90 days ago) https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/LibUsbDotNet/Generated/Device.ReleaseHandle.cs#L30

I'm not sure I can fix the without understanding when and why this is called.

Also I checked out the master repo at be7914ce0819748ea1f60728267f83efd97c1f6e built by project and everything ran as expected no crash on Win10. So something between be7914ce0819748ea1f60728267f83efd97c1f6e and latest commit causes the crash on Win10.

mcuee, 2023-07-14:
@CartBlanche 
Can you post a simple program which can trigger this issue?

Which version of dotnet are you using? Are you using the same dotnet version under Windows 10 and Windows 11?

sdijoseph, 2023-07-14:
> @mcuee, @sdijoseph The crash on Windows 10 happens in here (code was merged into the code base about 90 days ago) https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/LibUsbDotNet/Generated/Device.ReleaseHandle.cs#L30
> 
> I'm not sure I can fix the without understanding when and why this is called.
> 
> Also I checked out the master repo at [be7914c](https://github.com/LibUsbDotNet/LibUsbDotNet/commit/be7914ce0819748ea1f60728267f83efd97c1f6e) built by project and everything ran as expected no crash on Win10. So something between [be7914c](https://github.com/LibUsbDotNet/LibUsbDotNet/commit/be7914ce0819748ea1f60728267f83efd97c1f6e) and latest commit causes the crash on Win10.

If I'm understanding you correctly, I can make a few guesses:

1. The crash you are experiencing was introduced in PR #193 
2. If the crash happens at Device.ReleaseHandle, the device is being double de-referenced.
3. I'm not sure why this would happen on Windows 10 but not Windows 11.

CartBlanche, 2023-07-19:
@sdijoseph Apologies for the late reply. I was wrong, going back to a previous issue does fix the issue. As pointed out above during debugging the code base it steps through beautifully, but build the change in Release mode and it blows up fairly consistently. So i agree with @hellow554  that it seems to be a timing/ race condition issue.

In my case I'm using dfu-util to move the OS into place,  while in bootloader mode. The device is then reset to normal mode before using LibUSBDotNet to update our mono runtime on device. So that involves a `device.Open()` call.

CartBlanche, 2023-08-03:
@mcuee, @sdijoseph Any additional news about this issue on Windows 10?

@hellow554 Which dotnet sdks do you have installed when you see this issue?

hellow554, 2023-08-04:
@CartBlanche I'm not quiet sure what you mean, but the `Microsoft.Windows.SDK.NET` assembly has the version `10.0.22621.24` if you need that

sdijoseph, 2023-08-05:
> I have the exact same issue **on Windows 11**, here's my code, taken from [the example](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/019195538380576d9fb6eaefddf88c9b459ac47d/src/Examples/Show.Info/ShowInfo.cs#L13C20-L20):
> 
> ```cs
> using (var ctx = new UsbContext()) {
>     var allDevices = ctx.List();
>     foreach (var device in allDevices) {
>         bool opened = device.TryOpen();
>         Debug.WriteLine(String.Format("{0:X4}.{1:X4}", device.Info.VendorId, device.Info.ProductId));
>     }
> }
> ```
> 
> It really seems to be a timing issue, because if I step through it with a debugger, everything is fine, but as soon as I let it run without one it crashes on ~the first `Info` access.~ Actually it's not the first, but it seems that it's the last one and always the last one.
> 
> Happens with the latest 102 and also your 122 version.

I can't replicate this on Windows 11 with current git master. What version of libusb are you using? I'm using the latest release, [v1.0.26](https://github.com/libusb/libusb/releases/tag/v1.0.26). You can print the version of libusb that LibUsbDotNet is resolving using this [property](https://github.com/LibUsbDotNet/LibUsbDotNet/blob/019195538380576d9fb6eaefddf88c9b459ac47d/src/LibUsbDotNet/LibUsb/NativeLibraryVersion.cs#L68C26-L68C34).

CartBlanche, 2023-10-04:
@sdijoseph It mainly manifests itself on quite a lot of Windows 10 machines, but only some Windows 11 machines. we still don't know exactly why.

CartBlanche, 2023-10-04:
I think this new issue - https://github.com/LibUsbDotNet/LibUsbDotNet/issues/225 is a variation on the same problem.

mcuee, 2023-10-07:
@sdijoseph

Just wondering if you can take a look at this issue and #225, when you got some time. Thanks.

mcuee, 2023-10-18:
> @mcuee I think this newly reported issue - #225 is a variation on the same problem. To re-iterate it happens mostly on Win10 (again not all), and some Win11 (much less that Win10).

@CartBlanche 

Interestingly that #225 has been closed by the reporter and it can be sorted by upgraded to .Net 7.
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/225#issuecomment-1761708048

Can you try .Net 7 build to see if that helps to sort the issue for you? Thanks.

mcuee, 2025-04-09:
Just wondering if the latest version helps or not. Thanks.
https://www.nuget.org/packages/LibUsbDotNet/3.0.167-alpha#versions-body-tab