LibUsbDotNet/LibUsbDotNet #221: System.NullReferenceException while invoking LibUsbDotNet.LudnMonoLibUsb.MonoUsbDevice.OpenEndpointWriter

pavanupb, 2023-07-17:
I have an application where I am opening a connection to an instrument from my .Net 6.0 application using the static method `OpenUsbDevice` from the class UsbDevice by passing in the vendor and product Id to UsbDeviceFinder. I am using the same object to push the message by calling the `OpenEndPointWriter((WriteEndpointID)<output_pin>)`. However from time to time I am getting a NullReference exception. Below is the stack trace:

```
 System.NullReferenceException: Object reference not set to an instance of an object.
         at LibUsbDotNet.LudnMonoLibUsb.MonoUsbDevice.OpenEndpointWriter(WriteEndpointID writeEndpointID, EndpointType endpointType)
        at Trimble.MonSol.InstrumentManager.Sdk.Communicators.Linux.UsbCommunicatorWrapper.PushMessage(Byte[] message)
```

In my logs I am able to see that the `IsPortOpen` is true and was able to see the information as well from the UsbDevice object at the time of exception.

`Version: LibUsbDotNet 2.2.29`

Please let me know If I am doing something wrong

mcuee, 2023-07-17:
Can you try the latest v2 git branch? 2.2.29 has quite some issues.

mcuee, 2023-07-17:
Or you may want to try the latest master branch (V3), some changes may be required if you switch to V3.

pavanupb, 2023-07-17:
> Can you try the latest v2 git branch? 2.2.29 has quite some issues.

@mcuee Thanks for the suggestion. Could you please let me know which version in nuget are related to v2. I can see a list of alpha versions in nuget gallery

pavanupb, 2023-07-17:
@mcuee Is the artifacts from v2 branch pushed to nuget gallery?

pavanupb, 2023-07-17:
Thanks @mcuee . Are there any plans to release the v2 version to the nuget in near future?

mcuee, 2023-07-17:
> Thanks @mcuee . Are there any plans to release the v2 version to the nuget in near future?

We have to wait @qmfrederik to do that.

closed 2023-09-21 by mcuee

mcuee, 2026-01-22:
> [@mcuee](https://github.com/mcuee) [@sdijoseph](https://github.com/sdijoseph) When can we expect a new official release of v3 and publishing the package to nuget?. I saw the post: [#266](https://github.com/LibUsbDotNet/LibUsbDotNet/issues/266). Currently I am using 3.0.154-alpha.

Have you tried the latest version?
https://www.nuget.org/packages/LibUsbDotNet/#versions-body-tab
https://www.nuget.org/packages/LibUsbDotNet/3.0.167-alpha