LibUsbDotNet/LibUsbDotNet #223: 3.0.102-alpha memory error

zhangxinling2, 2023-07-27:
[image]
my problem like that,but something is different
This error occurs when I finish a write, but if I add a delay of a few seconds at the end it doesn't appear，this is why，how can i solve

mcuee, 2023-07-27:
Please do not use libusbdotnet with USB HID device.

Please use HIDAPI and its dotnet bindings.
https://github.com/libusb/hidapi/wiki

mcuee, 2023-07-27:
BTW, your read endpoint (IN Endpoint) should be 0x81 and not 0x01.

closed 2023-07-27 by mcuee

zhangxinling2, 2023-07-28:
[image]
I successfully operated with USB using libusb via QT, but now I need to use C# to operate on usb。
In fact, this  write data successfully, I finded changes to USB devices, but an memory error occurs after the write succeeds.
Strangely, I don't get this error by adding a sleep after a successful write
I think libusbdotnet worked, but don't know why the error happened

zhangxinling2, 2023-07-28:
> BTW, your read endpoint (IN Endpoint) should be 0x81 and not 0x01.

[image]
I successfully operated with USB using libusb via QT, but now I need to use C# to operate on usb。 In fact, this write data successfully, I finded changes to USB devices, but an memory error occurs after the write succeeds. Strangely, I don't get this error by adding a sleep after a successful write I think libusbdotnet worked, but don't know why the error happened

mcuee, 2023-07-31:
Have you tried git master and not 3.0.102-alpha?

zhangxinling2, 2023-07-31:
> Have you tried git master and not 3.0.102-alpha?

I use 
IUsbDevice usbDevice = (UsbDevice)context. Find(MyUsbFinder); 
instead of
 var usbDeviceCollection = context. List();
 var usbDevice = usbDeviceCollection.FirstOrDefault(d => d.ProductId == pid && d.VendorId == vid); 
This error no longer occurs.