LibUsbDotNet/LibUsbDotNet #224: Issue detecting device 0403:6001

RobertoBiundo, 2023-08-07:
0403:6001 is an FTDI USB to UART dongle. 

running:
`UsbDevice.AllDevices.ToList();`

Does make the device show properly in Linux and one can get the device serianNumber, etc.

However in windows this same code returns nothing (empty array) since there are no other USB device connected besides the UART dongle.

The device does show up in the device Manager as a USB device as well as a COM device.

What could be the problem here?

mcuee, 2023-08-07:
Under Windows, in order to use libusb/libusbdotnet, you have to use a supported driver.
https://github.com/libusb/libusb/wiki/Windows#how-to-use-libusb-on-windows

Which version of libusbdotnet are you using? Do you have libusb-1.0.dll in the path?

RobertoBiundo, 2023-08-09:
I'm using libusb 2.2.29
Unfortunately using a custom driver would cause windows to stop displaying the device a a serial port device as well. 
All I want in this case is to be able to get the serial number if the device

mcuee, 2023-08-09:
In that case, libusbdotnet is not the right library for you.

mcuee, 2023-08-09:
There is a PR for libusb which has not been merged.
* https://github.com/libusb/libusb/pull/1258

If that is merged, it is possible to add that to libusbdotnet main line (3.x version).

closed 2023-08-09 by mcuee

mcuee, 2023-08-09:
@RobertoBiundo 

You may want to look at FTDI's D2XX API.
https://ftdichip.com/software-examples/code-examples/csharp-examples/

mcuee, 2023-08-09:
If you need cross platform library support, then you may want to look at libserialport. I am not so sure if it has a dotnet binding or not.
http://sigrok.org/wiki/Libserialport

RobertoBiundo, 2023-08-09:
I manager to do this. 
Here is how
In Linux
1. Call libusbdotnet (the device will show up)
2. Get the serial number of the device
3. Check on /dev/tty* for tty devices and filter by the serial number you got from the libusb.
4. You now have a match

On windows
1. Use the managed components API from Microsoft to get all USB devices.
2. Using the serial number of the device use the managed.components again searching by COM.
3.  Get the serial from the COM and match it with the serial from the USB call
4. You now have a match

The restriction is you can't send messages to the endpoint under windows but the process meets my use case of having a generic code to match HID and non HID devices cross platform.
They are both recognized using the same library now but when talking with non HID it always falls back to UART