LibUsbDotNet/LibUsbDotNet #225: libusbdotnet 3.0.102 "Attempted to read or write protected memory. This is often an indication that other memory is corrupt."

hasann92, 2023-08-08:
I have one form application that is using libusbdotnet 3.0.102 aplha. when i open the connection and close the form most of the time gives an error that says "'Attempted to read or write protected memory. This is often an indication that other memory is corrupt.'"  Thanks in advance

HERE IS THE Code where the error returns 

protected override bool ReleaseHandle()
		{

			NativeMethods.Close(handle);
			return true;
		}

Here is my code: 

 public USBCom()
        {
            context = new UsbContext();
            usbFinder = new UsbDeviceFinder(vendorId, productId);
            usbDevice = context.Find(usbFinder);

            // Initialize the timer
            deviceCheckTimer = new System.Timers.Timer(500); // Check every 500 msec
            deviceCheckTimer.Elapsed += CheckDeviceConnection;
            deviceCheckTimer.Start();

            // Attempt initial connection
            ConnectToDevice();
        }

        public static USBCom Instance
        {
            get
            {
                lock (_lock)
                {
                    if (_instance == null)
                    {
                        _instance = new USBCom();
                    }
                    return _instance;
                }
            }
        }

        private void ConnectToDevice()
        {
            try
            {
                usbDevice = context.Find(usbFinder);
                if (usbDevice != null)
                {
                    usbDevice.Open();
                    usbDevice.SetConfiguration(1);
                    usbDevice.ClaimInterface(0);
                    USBinfo = usbDevice.Info;
                    isDeviceConnected = true;
                    DeviceConnected?.Invoke(this, EventArgs.Empty);
                }
            }
            catch (Exception ex)
            {
                MessageBox.Show(ex.Message);
            }
        }

hasann92, 2023-08-09:
Not sure how to do that. I added the library via nugget.

mcuee, 2023-08-09:
> Not sure how to do that. I added the library via nugget.

You have to build the library from source code.

hasann92, 2023-08-10:
Unfortunately, The same error. It does not affect the exe if I start the application from exe. It stops there whenever I run the code from the visual studio.

CartBlanche, 2023-10-19:
@hasann92 could you please confirm what .NET 7 sdks and runtimes you have on your machine to fix this, because I appear to have .NET 7 on my machine, but am still hitting this issue.

hasann92, 2023-12-21:
Sorry I just saw your comment. Could you please make sure  the usbcontext singleton instead creating multiple object? I have .net6 .net7 .net8 runtime installed I use net7. Whenever you call the usb object "context = new UsbContext();" use the only one context. Besides,  that has to be disposed when the application is closed

JKingas, 2025-03-25:
I also got this exception. Described it here https://github.com/LibUsbDotNet/LibUsbDotNet/issues/219#issuecomment-2751383900

Need to guaranteed dispose any `IUsbDevice` before dispose `UsbContext`.
Even if you use one `UsbContext`, and it is disposed before the `IUsbDevice` object, then there will be an exception again. That's why need to control `IUsbDevice` disposing in this version.

mcuee, 2025-04-08:
@JKingas 
Would you please try the latest version to see if that helps? Thanks.
https://www.nuget.org/packages/LibUsbDotNet/3.0.167-alpha#versions-body-tab