LibUsbDotNet/LibUsbDotNet #226: Does not find the device

Ovinov84, 2023-08-22:
Imports System.Collections.ObjectModel
Imports System.Reflection.Emit
Imports LibUsbDotNet
Imports LibUsbDotNet.Info
Imports LibUsbDotNet.Main
Module Program

    Private MyUsbDevice As UsbDevice
    Private MyUsbFinder As New UsbDeviceFinder(&H_04A9, &H_2736)
    Sub Main(args As String())
        MyUsbDevice = UsbDevice.OpenUsbDevice(MyUsbFinder)
        If MyUsbDevice IsNot Nothing Then
            Console.WriteLine("Подключено.")
        Else
            Console.WriteLine("Не найдено.")
        End If
    End Sub
End Module

mcuee, 2023-08-23:
You do need to provide more details about the OS, the device and the version of libusbdotnet used, as well as the driver used if you are using Windows.

Ovinov84, 2023-08-23:
1. Windows 10 Pro 21H2
2. Printer Canon MF4550d
3. LibUsbDotNet 2.2.29
4. usbccgp.sys (if it's him)

this is an example code

Imports System.Collections.ObjectModel
Imports System.Reflection.Emit
Imports LibUsbDotNet
Imports LibUsbDotNet.Info
Imports LibUsbDotNet.Main
Module Program

    Private MyUsbDevice As UsbDevice
    Private MyUsbFinder As New UsbDeviceFinder(&H_4E53, &H_5407)
    ' VID = &H_04A9, PID = &H_2736 - Принтер Canon
    Sub Main(args As String())
        MyUsbDevice = UsbDevice.OpenUsbDevice(Function(d) d.Vid = &H_04A9 And d.Pid = &H_2736)
        If MyUsbDevice IsNot Nothing Then
            Console.WriteLine("Подключено.")
        Else
            Console.WriteLine("Не найдено.")
        End If
    End Sub
End Module

P.S. I tried another code from the documentation on the site, he does not see any usb devices at all. What needs to be done?

closed 2023-09-21 by mcuee