LibUsbDotNet/LibUsbDotNet #227: System.DllNotFoundException:“Unable to load DLL 'libusb-1.0' or one of its dependencies

huster-songtao, 2023-09-20:
System.DllNotFoundException:“Unable to load DLL 'libusb-1.0' or one of its dependencies: 找不到指定的模块。 (0x8007007E)”

Unable to load DLL 'libusb-1.0' or one of its dependencies: 找不到指定的模块。 (0x8007007E)

   在 LibUsbDotNet.NativeMethods.Init(IntPtr& ctx)
   在 LibUsbDotNet.LibUsb.UsbContext..ctor() 在 C:\Users\Steven\Downloads\LibUsbDotNet-master\LibUsbDotNet-master\src\LibUsbDotNet\LibUsb\UsbContext.cs 中: 第 87 line
   在 Examples.ReadPolling.Main(String[] args) 在 C:\Users\Steven\Downloads\LibUsbDotNet-master\LibUsbDotNet-master\src\Examples\Read.Only\ReadOnly.cs 中: 第 23 line

mcuee, 2023-09-21:
You have to have `libusb-1.0.dll` in your system.

closed 2023-09-21 by mcuee

huster-songtao, 2023-09-21:
[image]

```
<Content Include="$(VCPKG_ROOT)/installed/x64-windows/bin/libusb-1.0.dll">
      <PackagePath>runtimes/win-x64/native/%(Filename)%(Extension)</PackagePath>
      <Pack>true</Pack>
    </Content>
```

To add the vcpkg directory to the environment variable: 
Environment variable name: VCPKG_ROOT 
Value: The installation directory of vcpkg, my path is E:\vcpkg\vcpkg

I have libusb-1.0.dll in E:\vcpkg\vcpkg\installed\x64-windows\bin

so, I have libusb-1.0.dll  in $(VCPKG_ROOT)/installed/x64-windows/bin/libusb-1.0.dll

mcuee, 2023-09-21:
That will not work. vcpkg will let Visual Studio find it but not your application. 

You can put `libusb-1.0.dll` in the same directory of your exe file.

huster-songtao, 2023-09-21:
[image]

To view a file's properties, right-click on the file and select "Properties" to open the Properties window below. 
[image]

the full path 'C:\installed\arm64-windows\bin\libusb-1.0.dll'
$(VCPKG_ROOT) is 'C:\' but not 'E:\vcpkg\vcpkg'

I have no idea for $(VCPKG_ROOT)
I will try later

mcuee, 2023-09-21:
Just ignore that. Put libusb-1.0.dll in the same directory as your exe file.