LibUsbDotNet/LibUsbDotNet #230: New nugut packages

mcuee, 2023-10-07:
From here.
* https://github.com/LibUsbDotNet/LibUsbDotNet/pull/216#issuecomment-1729607709

> @qmfrederik There may be some refinements before v3 is officially released, but this feature is good enough for a pre-release version for now. I've been using it in my applications for a few months now without issue.
> 
> Thanks for approving the PR, but please also trigger another NuGet build so people can easily use the new features that have been added in the last few months (I would also suggest adding @mcuee as a package owner if possible). Also, please follow the steps [here](https://docs.github.com/en/pages/quickstart#creating-your-website) (no. 3 onwards, "gh-pages" should be the branch) to deploy the documentation site added in #211 .

mcuee, 2023-10-07:
@qmfrederik 

Please help to trigger the build of nuget for both V3 (maybe we can call it beta release now) and V2.

mcuee, 2023-10-18:
For those who want to test out, please try the snapshot release.
https://github.com/LibUsbDotNet/LibUsbDotNet/releases/tag/snapshot_22Sept2023 (main branch)
https://github.com/LibUsbDotNet/LibUsbDotNet/releases/tag/snapshot_21Sept_2023_v2 (legacy v2 branch)

cristianst85, 2023-12-14:
Hello. Is there any chance to publish a new NuGet package for version 2.x? The last one (2.2.29) was published five years ago. Thank you.

mcuee, 2023-12-15:
> Hello. Is there any chance to publish a new NuGet package for version 2.x? The last one (2.2.29) was published five years ago. Thank you.

We have to wait for @qmfrederik as only he has the rights.

mcuee, 2024-05-05:
@qmfrederik 

Ping. Thanks.

beachcitiessoftware, 2024-07-30:
Is there any possibility of getting the latest changes for .NET 8.0 and async published to NuGet?   I've tried just adding this project as a git subrepo, but am getting the following error, when trying to connect:

Unable to load DLL 'libusb-1.0' or one of its dependencies: The specified module could not be found. 

I have installed the latest drivers from mcuee using zadig-2.9.exe.  

Any suggestions?

mcuee, 2024-11-17:
@qmfrederik 

Sorry to ping you again. Thanks.

mcuee, 2024-12-12:
For users of the legacy v2 version, you may want to check out the following fork.
https://www.nuget.org/packages/Huddly.LibUsbDotNet
https://github.com/Huddly/LibUsbDotNet/

mcuee, 2024-12-12:
@georgehahn Just wondering if you can help to upload new version of libusbdotnet nuget. Thanks.
https://www.nuget.org/packages/LibUsbDotNet/

GeorgeHahn, 2024-12-12:
@mcuee Sorry, I don't have appropriate context to publish new releases here. I'm not a member of this repo and I haven't worked with this library in a long time.

I have a proposal for a path forward. If you make an issue and add a readme notice alerting users of 'intent to transfer control of releases' and no objections are raised within two weeks, I will add you to the Nuget project and remove myself. How does that sound?

mcuee, 2024-12-13:
@GeorgeHahn 

Thanks a lot for the help.
Issue #258 has been raised.
* #258

mcuee, 2025-01-09:
I have got the owner right and #258 has been closed.

I will try to resolve this issue in the near future.

mcuee, 2025-02-19:
Need to fix the following issue first.
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/250

Is it a good idea to ask the user to get libusb-1.0 binary by themselves?

mcuee, 2025-02-19:
> Is it a good idea to ask the user to get libusb-1.0 binary by themselves?

@sdijoseph 
Just wondering what is your opinion. Thanks.

sdijoseph, 2025-02-19:
> > Is it a good idea to ask the user to get libusb-1.0 binary by themselves?
> 
> [@sdijoseph](https://github.com/sdijoseph) Just wondering what is your opinion. Thanks.

That seems reasonable to me, we can just leave instructions on how to download/build the binary in place of the binary itself.

mcuee, 2025-02-20:
@sdijoseph 
Thanks a lot for the help.

Just wondering if you can help to write a short guide for me to create a proper nuget package from the git repo. Thanks. I have not done that before and I am afraid to do things wrongly.

If you are okay, I can add you to be the owner of the nuget package.

sdijoseph, 2025-02-21:
> [@sdijoseph](https://github.com/sdijoseph) Thanks a lot for the help.
>
> Just wondering if you can help to write a short guide for me to create a proper nuget package from the git repo. Thanks. I have [...]

I'd love to help, but it will probably have to wait until April 1st, I'm very busy until then.

sdijoseph, 2025-04-06:
> [@sdijoseph](https://github.com/sdijoseph) Thanks a lot for the help.
>
> Just wondering if you can help to write a short guide for me to create a proper nuget package from the git repo. Thanks. I have [...]

Hi, I can try to put together a NuGet release pipeline if you add me as a package owner now.

mcuee, 2025-04-06:
> Hi, I can try to put together a NuGet release pipeline if you add me as a package owner now.

Please give me your user namer at Nuget.org. Thanks.

sdijoseph, 2025-04-06:
> > Hi, I can try to put together a NuGet release pipeline if you add me as a package owner now.
> 
> Please give me your user namer at Nuget.org. Thanks.

sdijoseph

mcuee, 2025-04-06:
@sdijoseph 

Great. I have sent you the invitation.

sdijoseph, 2025-04-08:
> @sdijoseph 
> 
> Great. I have sent you the invitation. 
> 

Do you have the ability to add a user secret to the repo? I believe we will need it to add the nuget API key to the GitHub actions workflow.

https://docs.github.com/en/actions/security-for-github-actions/security-guides/using-secrets-in-github-actions#creating-secrets-for-a-repository

mcuee, 2025-04-08:
> Do you have the ability to add a user secret to the repo? I believe we will need it to add the nuget API key to the GitHub actions workflow.
> 
> https://docs.github.com/en/actions/security-for-github-actions/security-guides/using-secrets-in-github-actions#creating-secrets-for-a-repository

Unfortunately I do not have the owner right to change any settings of this repo. 

Is  it possible to manually upload buget packages to nuget.org?

sdijoseph, 2025-04-08:
> Unfortunately I do not have the owner right to change any settings of this repo.
>
> Is it possible to manually upload buget packages to nuget.org?

Yes, for now I think I can just push the latest artifact .nupkg that was built from master. Is that alright?

mcuee, 2025-04-08:
> > Unfortunately I do not have the owner right to change any settings of this repo.
> > Is it possible to manually upload buget packages to nuget.org?
> 
> Yes, for now I think I can just push the latest artifact .nupkg that was built from master. Is that alright?

Yes, please. Thanks a lot for the help.

sdijoseph, 2025-04-08:
> Yes, please. Thanks a lot for the help.

Ok, I uploaded the latest build of master: https://www.nuget.org/packages/LibUsbDotNet/3.0.167-alpha#versions-body-tab

mcuee, 2025-04-08:
@sdijoseph 
Great. Thanks a lot for the help.

Just wondering if you can do the same for legacy v2 branch as well, probably just one last time.

sdijoseph, 2025-04-12:
> [@sdijoseph](https://github.com/sdijoseph) Great. Thanks a lot for the help.
> 
> Just wondering if you can do the same for legacy v2 branch as well, probably just one last time.

v2 package uploaded - https://www.nuget.org/packages/LibUsbDotNet/2.2.75

closed 2025-04-12 by mcuee

mcuee, 2025-05-08:
@sdijoseph 

Just wondering what is the git commit corresponding to the above nuget package.

v3 package: latest git, https://github.com/LibUsbDotNet/LibUsbDotNet/commit/f846f1607a1931733e7fc3ba5870b5d7e6e05e51

v2 package 2.2.75: https://github.com/LibUsbDotNet/LibUsbDotNet/commit/2d7289ab5d4059ac9bfbde666cfb0d4825d89814