LibUsbDotNet/LibUsbDotNet #231: Getting System.NullReferenceException when trying to disconnect

pavanupb, 2023-11-10:
I am using the latest v2 branch nupkg file posted in the [](https://github.com/LibUsbDotNet/LibUsbDotNet/issues/230).
I have an application where I am opening a connection to an instrument from my .Net 6.0 application using the static method `OpenUsbDevice` from the class UsbDevice by passing in the vendor and product Id to UsbDeviceFinder. I am using the same object to push the message by calling the `OpenEndPointWriter((WriteEndpointID)<output_pin>)`. However from time to time I am getting a NullReference exception. Below is the stack trace:

```
 System.NullReferenceException: Object reference not set to an instance of an object.
         at LibUsbDotNet.LudnMonoLibUsb.MonoUsbDevice.OpenEndpointWriter(WriteEndpointID writeEndpointID, EndpointType endpointType)
***** App_Stack trace *******  
```

In my logs I am able to see that the `IsPortOpen` is true and was able to see the information as well from the UsbDevice object at the time of exception.

`Version: LibUsbDotNet 2.2.62 (Latest v2 branch)`

`OS: Raspi OS`

I am running my application inside a docker container on a Raspberry Pi 4 machine

Please let me know If I am doing something wrong

pavanupb, 2023-11-10:
@mcuee I had posted the same issue before. As suggested by you I tried with the latest V2 branch and seems that I am facing the same issue even after using the latest nupkg.

mcuee, 2023-11-10:
> @mcuee I had posted the same issue before. As suggested by you I tried with the latest V2 branch and seems that I am facing the same issue even after using the latest nupkg.

Since you are using Linux, please use the main branch V3 branch -- now it has proper hotplug support.

Main development will be on the main V3 branch.

sdijoseph, 2023-11-10:
> Thanks @mcuee. Could you please let me know if there is a documentation for the V3 branch?. I see that there are some breaking changes when I update to V3

The V3 version of the library is very different. I made a documentation page a while ago, but I'm waiting on the repo owner to add the deployment in the repo settings. In the meantime, you can use the page on my fork:

https://sdijoseph.github.io/LibUsbDotNet/api/index.html

However, also be aware the latest version of V3 in git is not packaged in NuGet yet on the main feed. We're also waiting on the repo owner to release a new package version.

pavanupb, 2023-11-13:
Thanks @sdijoseph. I am trying to modify my code to align with the new V3 syntax. I just noticed that the dispose on UsbEndpointWriter and UsbEndpointReader is removed, instead UsbContext is used. I wanted to check is it always recommended to open a UsbContext and then perform operations like close, read, write, etc. I have different methods for disconnect, read and write. Before I would dispose them individually when the operation is complete. Now should I open the UsbContext, FindDevice(based on Pid and Vid) and then for eg: open the reader, writer or close the connection?

pavanupb, 2023-11-14:
@mcuee I am getting: ```
SafeHandle cannot be null. (Parameter 'pHandle')
edge                 | System.ArgumentNullException: SafeHandle cannot be null. (Parameter 'pHandle')
edge                 |    at System.StubHelpers.StubHelpers.SafeHandleAddRef(SafeHandle pHandle, Boolean& success)
edge                 |    at LibUsbDotNet.NativeMethods.BulkTransfer(DeviceHandle devHandle, Byte endpoint, Byte* data, Int32 length, Int32& actualLength, UInt32 timeout)
edge                 |    at LibUsbDotNet.LibUsb.UsbEndpointBase.Transfer(Span`1 buffer, Int32 offset, Int32 length, Int32 timeout, Int32& transferLength)
edge                 |    at LibUsbDotNet.LibUsb.UsbEndpointReader.Read(Byte[] buffer, Int32 offset, Int32 count, Int32 timeout, Int32& transferLength)
edge                 |    at LibUsbDotNet.LibUsb.UsbEndpointReader.Read(Byte[] buffer, Int32 timeout, Int32& transferLength)
```
while trying to open the reader.
I was able to run with V2 but when I updated to V3 I am getting the above exception. Do I need to handle something differently?

sdijoseph, 2023-11-17:
> @mcuee I am getting:
> ```
> SafeHandle cannot be null. (Parameter 'pHandle')
> edge                 | System.ArgumentNullException: SafeHandle cannot be null. (Parameter 'pHandle')
> edge [...]

Can I see a snippet of your code?

pavanupb, 2023-11-20:
@sdijoseph Thanks. I fixed the issue. I was managing the UsbContext in a wrong manner.

closed 2025-04-08 by mcuee