LibUsbDotNet/LibUsbDotNet #232: Questions about HID devices

GreenShadeZhang, 2023-12-14:
I don't know if libusbdotnet has the ability to operate HID devices. I want to know if this library can add such capabilities, or if I can contribute something in this area, because I looked at a device.Net library and I found that the author It is no longer updated, so it would be helpful if this library could be made cross-platform and support HID.

mcuee, 2023-12-15:
libusb project recommends using HIDAPI for HID devices (USB, I2C, SPI, Bluetooth, BLE).

So it is better that you look at dotnet bindings of HIDAPI.
Eg:
https://github.com/badcel/HidApi.Net
https://github.com/alexgubanow/hidapiw

Reference:
https://github.com/libusb/libusb/wiki/FAQ#user-content-Does_libusb_support_USB_HID_devices
https://github.com/libusb/hidapi/wiki

GreenShadeZhang, 2023-12-15:
> libusb project recommends using HIDAPI for HID devices (USB, I2C, SPI, Bluetooth, BLE).
>
> So it is better that you look at dotnet bindings of HIDAPI. Eg: https://github.com/badcel/HidApi.Net [...]

[image]
Thank you very much. I see that the hidapi.net library is very new and its functions are also very complete. I will test this library. Is there any plan to support .net8 in the v3 version of libsubdotnet?

mcuee, 2023-12-15:
> Is there any plan to support .net8 in the v3 version of libsubdotnet?

.NET 8 support will be good to have.

PR will be welcome.

GreenShadeZhang, 2023-12-15:
> > Is there any plan to support .net8 in the v3 version of libsubdotnet?
> 
> .NET 8 support will be good to have.
> 
> PR will be welcome.

It sounds great. I wonder if the author forgot his github account. Regarding publishing the code to the nuget server, it seems that it has not been updated there for a long time. I really hope the author can give you publishing permission.

mcuee, 2023-12-22:
#235 has been raised to address the .NET 8 support enhancement request.

closed 2023-12-22 by mcuee