LibUsbDotNet/LibUsbDotNet #233: Class 'NativeMethods' is inaccessible due to its protection level

mikunal, 2023-12-15:
Can you pls make **NativeMethods** class public?
(https://github.com/LibUsbDotNet/LibUsbDotNet/blob/master/src/LibUsbDotNet/Generated/NativeMethods.cs)
**_internal_** static unsafe partial class **NativeMethods**

Often, it is important to have both:  access to "libusb-1.0" functionality via high-level abstractions, as well as via direct low-level API calls.

mcuee, 2023-12-22:
Is this the same idea as #125?
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/125

mikunal, 2023-12-22:
#125 is a MUCH bigger request. Mine is _simply changing_ '**internal'** to **'public'** in a single file.
ZERO other changes. No extra testing needed

mcuee, 2023-12-22:
@sdijoseph 

Please help to comment on this idea. Thanks.

mikunal, 2023-12-22:
Reasons for having **_internal_** **static unsafe partial class NativeMethods** (instead of **_public_** **static unsafe partial class NativeMethods**) are rather unclear to me. Enabling external/direct access to P/Invoke methods seems like a natural thing to do.

sdijoseph, 2024-01-05:
I think the appeal of LibUsbDotNet is that it is a managed wrapper for libusb; directly calling the native methods seem to defeat the purpose of using the library at all.  There is a high chance that mixing managed and native calls will cause difficult to debug bugs like memory corruption (I ran into a few of these when adding async + hotplug features). I would have zero interest in looking into any future issue raised by someone using the native methods directly in their code. If they are made public, it should probably come with a warning that it's "use at your own risk".

I'm also interested in what functionality you want from the native API that isn't available in the managed one. @mikunal 

That's just my opinion though, really I think @qmfrederik should make this call.

mcuee, 2024-05-05:
Close this issue as not planned for now.

closed 2024-05-05 by mcuee