LibUsbDotNet/LibUsbDotNet #234: LibUSB Read Port Number

hasann92, 2023-12-21:
I amdeveloping a C# application that needs to manage USB devices using the LibUsbDotNet library. Specifically, you require a way to identify when a USB device is disconnected from the system. Ideally, I want to use a numerical identifier for each device, such as a "port number," because serial numbers are not reliably available due to devices being open and in use by other processes, which prevents access to their serial number.  In debugging the UsbDevice object, I 've identified a PortNumber property that appears to hold the information I need. However, this property is not accessible in my code, likely because it is non-public. My goal is to access the value of the PortNumber property to track the device's connection status.

I began by attempting to access the PortNumber property directly in your code, which resulted in a compile-time error indicating that the UsbDeviceInfo class does not contain a definition for PortNumber.

To circumvent the direct access issue, I  attempted to use reflection to access the PortNumber property. This initially involved checking for both a field and a property with the name PortNumber using reflection with appropriate BindingFlags.

[image]

mcuee, 2023-12-22:
#192 has been merged in libusbdotnet git. Please look at the example there to see it works for you or not. Thanks.
* https://github.com/LibUsbDotNet/LibUsbDotNet/pull/192

hasann92, 2023-12-22:
Thank you so much. I had to delete and reinstall the package beside instead of using "var" I have changed to "UsbDevice" as the suggested link shows.

` var currentDevices = _context.List().Where(d => d.Info.VendorId == VENDOR_ID && PRODUCT_ID.Contains(d.Info.ProductId)).ToList();
 
 if (currentDevices.Count < USBDevices.Count)
 {     
     foreach (UsbDevice currentDevice in currentDevices)
     {
         var mk= currentDevice.PortNumber;
     }`

closed 2023-12-22 by hasann92