LibUsbDotNet/LibUsbDotNet #253: The README suggestion to add libusb-1.0.so symlink under Linux is wrong

zboszor, 2024-10-22:
The `.so` symlink is strictly speaking part the SDK, that is to allow compiling and linking to the libusb-1.0 library.

Instead, a library mapping is needed in `/etc/mono/config` to use the production library name, like in this PR: https://github.com/mono/mono/pull/21807

mcuee, 2024-10-22:
Down the road, Mono may not be the focus or even supported. 

The focus is the official Microsoft .NET.

zboszor, 2024-10-22:
Still, recommending the `.so` **SDK** symlink is not applicable for production machines. Development machines already have it but we want to run C# code in production. A similar mapping must be provided in .NET.

zboszor, 2024-10-22:
I think it would be better to change the source code instead, so neither the .so SDK symlink nor the Mono-specific mapping change are needed.

See [how NNanomsg does it](https://github.com/d-remeika/NNanomsg/blob/master/NNanomsg/Interop.cs): the library name is constructed during library initialization, instead of using static code.

mcuee, 2024-10-22:
@sdijoseph

Please help to comment when you got the time. Thanks.

madewokherd, 2024-10-22:
For Mono, the correct solution would be to include a .dll.config file with your assembly that contains the dllmap. See https://www.mono-project.com/docs/advanced/pinvoke/

mcuee, 2025-01-17:
I have removed the outdated info in https://github.com/LibUsbDotNet/LibUsbDotNet/commit/c8153e3145db92373af7466baf36d65441d19038.

The idea is to update Wiki to help the users.

closed 2025-04-08 by mcuee