LibUsbDotNet/LibUsbDotNet #258: Intent to transfer control of releases

mcuee, 2024-12-13:
As of now, libusbdotnet nuget package owners are @qmfrederik and @GeorgeHahn.
https://www.nuget.org/packages/LibUsbDotNet/

Unfortunately @qmfrederik is quite busy. @GeorgeHahn mentioned that he had not worked with this project for a long time. He has the following proposal to move things forward.

https://github.com/LibUsbDotNet/LibUsbDotNet/issues/230#issuecomment-2539967386
> @mcuee Sorry, I don't have appropriate context to publish new releases here. I'm not a member of this repo and I haven't worked with this library in a long time.
> 
> I have a proposal for a path forward. If you make an issue and add a readme notice alerting users of 'intent to transfer control of releases' and no objections are raised within two weeks, I will add you to the Nuget project and remove myself. How does that sound?

mcuee, 2024-12-13:
@qmfrederik 
Please help to add your comment if you have the time. Thanks.

mcuee, 2024-12-13:
For all libusbdotnet project users, please comment as well if you have any objections. Thanks.

GeorgeHahn, 2025-01-02:
@mcuee Please share your Nuget username and I'll make the change.

mcuee, 2025-01-04:
> @mcuee Please share your Nuget username and I'll make the change.

Thanks a lot for the help. My Nuget username is `xiaofanc`.

GeorgeHahn, 2025-01-04:
Invite sent. All the best with LibUsbDotNet! Once you accept the invitation, please remove me as an owner of the package.

mcuee, 2025-01-04:
> Invite sent. All the best with LibUsbDotNet! Once you accept the invitation, please remove me as an owner of the package.

Thanks a lot for the help. Somehow I have not received the invitation yet. I will check tomorrow.

mcuee, 2025-01-07:
@GeorgeHahn 

Not so sure what happened but I have not received any invitations. Please help to send again. Thanks.

mcuee, 2025-01-09:
@GeorgeHahn 

Done. I have accepted the invitation and removed your name. Thanks a lot for the help.

closed 2025-01-09 by mcuee