LibUsbDotNet/LibUsbDotNet #266: Releasing final version of v3

sdijoseph, 2025-04-12:
I wanted to ask some questions about LibUsbDotNet's future:

1. What is needed to take v3 out of prerelease? 
2. Are there major features present in v2 that are missing in v3?
3. When v3 is released officially, should v2 support be dropped?

My opinion on point 3 is yes, as v3 is netstandard2.0 compatible, so I'm not sure what purpose keeping v2 around would serve.

mcuee, 2025-04-12:
My take.
1) We can do it now.
2) I do not see any major missing features in v3 now.
3) I agree. I am thinking just to release v2 nuget package the last time and then the support will be dropped. Bug fixes for v2 may still be merged if the PR does not seem to be affecting v2.

mcuee, 2025-04-12:
I have closed all the v2 related issues as "not planned".

mcuee, 2025-04-14:
@sdijoseph 

A few small things.
1) to remove Benchmarkcon as we will not migrate it to v3
2) #197 --> to be fixed with PR #268 
3) #220 --> lower priority, or just remove it
4) #209 if any thing to be done.

sTolyaR, 2025-12-12:
Hello all!
What about isochronous mode with transfer queue in v3? We've tried to add such a functionality with v2 but still have some trouble with it.

mcuee, 2026-02-09:
I have updated the version.json file and cleanup the github actions.

antoniovazquezblanco, 2026-03-03:
I would say it makes sense. You can add a local nuget source for it to be able to find the updated project and then it should build.

glopesdev, 2026-03-14:
@mcuee @antoniovazquezblanco thanks for your efforts pushing this to a final release!

Is there any reason why we can't just move the project to net8.0?

As for the build error, why not simply add directly a project reference to that project, similar to LibUsbDotnet.Tests?

Since everything is being built together that would ensure the correct code version is used at all times, rather than trusting a local or remote package source.

mcuee, 2026-03-14:
@glopesdev 

No issues to move to .Net 8.0,

PR is welcome.

mcuee, 2026-03-14:
I actually do not know much about dotnet. I am more of a tester. I worked with Travis Robinson on libusb-win32, libusbK and libusbdotnet as a tester and also technical support (more for libusb-win32 and libusbK, less for libusbdotnet as I do not know much about dotnet).

Travis still keeps developing libusbK. I am also an admin.
https://github.com/mcuee/libusbk

Peter Dons Tychsen took over libusb-win32 project. I am also an admin.
https://github.com/mcuee/libusb-win32

libusbdotnet project was handed over to @qmfrederik later. Unfortunately he is too busy and not responding for a while. In the case of libusbdotnet, I have the code commit right to the repo but I am not an admin to this repo.

Since I still have the code commit right, I will try to test the PRs and merge the good ones (relying on the review of the community and my own testing).

mcuee, 2026-04-30:
Closer.

```
PS C:\work\libusb\libusbdotnet\libusbdotnet_v3> cd .\src\TestInfo\

PS C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo> dotnet add package Avalonia.Controls.DataGrid
info : X.509 certificate chain validation will use the default trust store selected by .NET for code signing.
info : X.509 certificate chain validation will use the default trust store selected by .NET for timestamping.
info : Adding PackageReference for package 'Avalonia.Controls.DataGrid' into project 'C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\TestInfo.csproj'.
info :   GET https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/index.json
info :   OK https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/index.json 424ms
info :   GET https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/page/0.8.0/11.0.0-preview1.json
info :   OK https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/page/0.8.0/11.0.0-preview1.json 258ms
info :   GET https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/page/11.0.0-preview2/11.3.11.json
info :   OK https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/page/11.0.0-preview2/11.3.11.json 235ms
info :   GET https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/page/11.3.12/12.0.0.json
info :   OK https://api.nuget.org/v3/registration5-gz-semver2/avalonia.controls.datagrid/page/11.3.12/12.0.0.json 252ms
info : Restoring packages for C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\TestInfo.csproj...
info :   CACHE https://api.nuget.org/v3-flatcontainer/avalonia.controls.datagrid/index.json
info :   GET https://api.nuget.org/v3-flatcontainer/avalonia.controls.datagrid/12.0.0/avalonia.controls.datagrid.12.0.0.nupkg
info :   OK https://api.nuget.org/v3-flatcontainer/avalonia.controls.datagrid/12.0.0/avalonia.controls.datagrid.12.0.0.nupkg 58ms
info : Installed Avalonia.Controls.DataGrid 12.0.0 from https://api.nuget.org/v3/index.json to C:\Users\XFCHEN\.nuget\packages\avalonia.controls.datagrid\12.0.0 with content hash pQ9NiviIoxbZ7fiD/KTdoEHRgdKibKtx8Low3nTrU6ZbvULtesuB65aLL/umjibsGMhzPOveqs/om98QAA5FMg==.
info :   CACHE https://api.nuget.org/v3/vulnerabilities/index.json
info :   CACHE https://api.nuget.org/v3-vulnerabilities/2026.04.28.03.00.19/vulnerability.base.json
info :   CACHE https://api.nuget.org/v3-vulnerabilities/2026.04.28.03.00.19/2026.04.30.03.00.30/vulnerability.update.json
error: NU1605: Warning As Error: Detected package downgrade: Avalonia from 12.0.0 to 11.2.7. Reference the package directly from the project to select a different version.
error:  TestInfo -> Avalonia.Controls.DataGrid 12.0.0 -> Avalonia (>= 12.0.0)
error:  TestInfo -> Avalonia (>= 11.2.7)
info : Package 'Avalonia.Controls.DataGrid' is compatible with all the specified frameworks in project 'C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\TestInfo.csproj'.
info : PackageReference for package 'Avalonia.Controls.DataGrid' version '12.0.0' added to file 'C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\TestInfo.csproj'.
info : Writing assets file to disk. Path: C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\obj\project.assets.json
log  : Failed to restore C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\TestInfo.csproj (in 483 ms).

PS C:\work\libusb\libusbdotnet\libusbdotnet_v3> git diff
diff --git a/src/TestInfo/TestInfo.csproj b/src/TestInfo/TestInfo.csproj
index a7e8ca4..b48d37b 100644
--- a/src/TestInfo/TestInfo.csproj
+++ b/src/TestInfo/TestInfo.csproj
@@ -17,6 +17,7 @@

   <ItemGroup>
     <PackageReference Include="Avalonia" Version="$(AvaloniaVersion)" />
+    <PackageReference Include="Avalonia.Controls.DataGrid" Version="12.0.0" />
     <PackageReference Include="Avalonia.Desktop" Version="$(AvaloniaVersion)" />
     <PackageReference Include="Avalonia.ReactiveUI" Version="$(AvaloniaVersion)" />
     <PackageReference Include="Avalonia.Themes.Fluent" Version="$(AvaloniaVersion)" />

PS C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo> dotnet build .\TestInfo.csproj
    C:\work\libusb\libusbdotnet\libusbdotnet_v3\src\TestInfo\TestInfo.csproj : error NU1605:
      Warning As Error: Detected package downgrade: Avalonia from 12.0.0 to 11.2.7. Reference the package directly from the project to select a different version.
       TestInfo -> Avalonia.Controls.DataGrid 12.0.0 -> Avalonia (>= 12.0.0)
       TestInfo -> Avalonia (>= 11.2.7)

Restore failed with 1 error(s) in 1.1s
```

mcuee, 2026-05-05:
@sdijoseph 

I understand that you are busy, but if you have some time, please take a look at the latest git as well. Thanks.

The target is to release the official release version of v3 by end of May 2026.

mcuee, 2026-05-07:
> So just release what is currently in master as the full v3 release? I think we should probably update the target frame works to netstandard2.0 and net10.0 since .NET 8 is longer the latest LTS version.

I see. I will prefer to release one version for .NET8 first.

Then we can bump to .NET10 in the next version, probably 4.0 series.

https://dotnet.microsoft.com/en-us/platform/support/policy/dotnet-core
Version | Original release date | Latest patch version | Patch release date | Release type | Support phase | End of support
-- | -- | -- | -- | -- | -- | --
.NET 10 | November 11, 2025 | 10.0.7 | April 21, 2026 | LTS | Active | November 14, 2028
.NET 9 | November 12, 2024 | 9.0.15 | April 14, 2026 | STS | Active | November 10, 2026
.NET 8 | November 14, 2023 | 8.0.26 | April 14, 2026 | LTS | Active | November 10, 2026

mcuee, 2026-05-11:
Now I will wait for another two weeks to see if there are further issues pop up. If now, I will release the current git as the official v3 release.

mcuee, 2026-05-17:
Make a few minor changes to use valid VID/PID in the examples.

Also commit the temporary workaround for the hotplug example. We can always wait for the proper fix.
* https://github.com/LibUsbDotNet/LibUsbDotNet/issues/295