LibUsbDotNet/LibUsbDotNet #270: No access to internal libusb_clear_halt

SlyCedix, 2025-06-18:
This method is required by, at least, the USBTMC specification during initialization and is currently inaccessible without `DllImport`ing the stub and calling it directly 

https://github.com/LibUsbDotNet/LibUsbDotNet/blob/f846f1607a1931733e7fc3ba5870b5d7e6e05e51/src/LibUsbDotNet/Generated/NativeMethods.cs#L188

It's a little hacky, but for anyone facing the same issue, this extension method allows you to call ClearHalt at the UsbDevice level with the correlated UsbEndpointInfo object:

```csharp
public static class NativeMethods
{
    [DllImport("libusb-1.0", EntryPoint = "libusb_clear_halt")]
    private static extern Error ClearHalt(DeviceHandle devHandle, byte endpoint);

    public static Error ClearHalt(this UsbDevice device, UsbEndpointInfo ep)
    {
        return ClearHalt(device.DeviceHandle, ep.EndpointAddress);       
    }
}
```

SlyCedix, 2025-06-30:
I could do a PR, what would the preferred API be. IMO the one I have above is a bit clunky because you may not actually have the `UsbEndpointInfo` in a context where you would be calling.

Take the byte directly:
```csharp
Error UsbDevice.ClearHalt(byte);
```

Or Overloads for each of the EndpointID enums
```csharp
Error UsbDevice.ClearHalt(ReadEndpointID);
Error UsbDevice.ClearHalt(WriteEndpointID);
```

mcuee, 2025-07-01:
@SlyCedix 

You have to specify the EndPoint as Clear Halt should target the endpoint and not the interface/device.

PR #271 (2025-07-09): Provide access to libusb_clear_halt
 * Adds `UsbEndpointBase.ClearHalt()`
Documentation from `man libusb`

Closes #270